mu-editor/mu #696: Mu-Editor suddenly CRASHED and ERASED the entire file

96konoha, 2018-10-31:
```
2018-10-30 23:18:26,254 - mu.logic:1138(autosave) INFO: Autosave detected and saved changes in /Users/konoha/Desktop/VIS142/project1/project1.py.
2018-10-30 23:18:29,414 - mu.logic:842(save_tab_to_file) INFO: Saving script to: /Users/konoha/Desktop/VIS142/project1/project1.py
2018-10-30 23:18:29,414 - mu.logic:843(save_tab_to_file) DEBUG: import pygame
from pygame.locals import *
import random

pygame.init()

SCREEN_SIZE = 900
screen = pygame.display.set_mode((SCREEN_SIZE, 120))

################################################################################
#                               Classes                                        #
################################################################################

class Alien(pygame.sprite.Sprite):
    
    def __init__(self, filename, x, y, index):
        super().__init__()
        self.rawimage = pygame.image.load(filename)
        self.image = self.rawimage
        self.rect = self.image.get_rect()
        self.rect.x = x
        self.rect.y = y
        self.index = index
        
    def get_rawimage(self):
        return self.rawimage
        
    def update(self):
        screen.blit(self.image, self.rect)

class Blur(pygame.sprite.Sprite):
    
    def __init__(self, rawimage):
        super().__init__()
        self.rawimage = rawimage
        self.image = self.rawimage
        self.rect = self.image.get_rect()
        self.rect.x = random.randrange(SCREEN_SIZE-100)
        self.rect.y = random.randrange(SCREEN_SIZE-100)
        self.angle = 0
        self.scale = 1
    
    '''
    def rotate(self, angle):
        self.image = pygame.transform.rotate(self.rawimage, angle)
    
    def scale(self, scale):
        self.image = pygame.transform.scale(self.rawimage, scale * self.rect.w, scale * self.rect.h)¶
    '''
    
    def rotozoom(self):
        self.angle += 5
        self.scale += 0.1
        self.image = pygame.transform.rotozoom(self.rawimage, self.angle, self.scale)
    
    def update(self):
        screen.blit(self.image, self.rect)
        if self.angle >= 90:
            self.kill()
    
    
alien_group = pygame.sprite.Group()
blur_group = pygame.sprite.Group()

for i in range(0,9):
    a = Alien('/Users/konoha/Desktop/VIS142/project1/alien.png', i*100+10, 10, i)
    alien_group.add(a)
[... 2152 lines cut ...]
2018-10-30 23:32:57,529 - root:114(run) INFO: Python path: ['/Applications/mu-editor.app/Contents/MacOS', '/Applications/mu-editor.app/Contents/Resources/python/lib/python36.zip', '/Applications/mu-editor.app/Contents/Resources/python/lib/python3.6', '/Applications/mu-editor.app/Contents/Resources/python/lib/python3.6/lib-dynload', '/Applications/mu-editor.app/Contents/Resources/python/lib/python3.6/site-packages', '/Applications/mu-editor.app/Contents/Resources/app', '/Applications/mu-editor.app/Contents/Resources/app_packages', '/Applications/mu-editor.app/Contents/Resources/app_packages/IPython/extensions']
2018-10-30 23:32:57,529 - root:115(run) INFO: Language code: en
2018-10-30 23:32:57,807 - mu.logic:540(__init__) INFO: Setting up editor.
2018-10-30 23:32:57,807 - mu.logic:558(__init__) INFO: Settings path: /Users/konoha/Library/Application Support/mu/settings.json
2018-10-30 23:32:57,807 - mu.logic:559(__init__) INFO: Session path: /Users/konoha/Library/Application Support/mu/session.json
2018-10-30 23:32:57,807 - mu.logic:560(__init__) INFO: Log directory: /Users/konoha/Library/Logs/mu
2018-10-30 23:32:57,807 - mu.logic:561(__init__) INFO: Data directory: /Users/konoha/Library/Application Support/mu
2018-10-30 23:32:57,823 - mu.logic:574(setup) INFO: Available modes: python, adafruit, microbit, debugger, pygamezero
2018-10-30 23:32:57,981 - mu.logic:1112(change_mode) INFO: Workspace directory: /Users/konoha/mu_code
2018-10-30 23:32:57,986 - mu.logic:623(restore_session) INFO: Restoring session from: /Users/konoha/Library/Application Support/mu/session.json
2018-10-30 23:32:57,986 - mu.logic:624(restore_session) DEBUG: {'theme': 'day', 'mode': 'python', 'paths': ['/Users/konoha/Desktop/VIS142/project1/project1.py'], 'envars': [], 'minify': False, 'microbit_runtime': ''}
2018-10-30 23:32:57,986 - mu.logic:708(_load) INFO: Loading script from: /Users/konoha/Desktop/VIS142/project1/project1.py
2018-10-30 23:32:57,987 - mu.logic:271(read_and_decode) DEBUG: Trying to decode with utf-8
2018-10-30 23:32:57,987 - mu.logic:274(read_and_decode) INFO: Decoded with utf-8
2018-10-30 23:32:57,987 - mu.logic:287(read_and_decode) DEBUG: Detected newline '\n'
2018-10-30 23:32:57,987 - mu.logic:782(_load) DEBUG: 
2018-10-30 23:32:58,077 - mu.logic:645(restore_session) INFO: Loaded files.
2018-10-30 23:32:58,077 - mu.logic:649(restore_session) INFO: User defined environment variables: []
2018-10-30 23:32:58,078 - mu.logic:653(restore_session) INFO: Minify scripts on micro:bit? False
2018-10-30 23:32:58,091 - mu.logic:1112(change_mode) INFO: Workspace directory: /Users/konoha/mu_code
2018-10-30 23:33:11,579 - mu.logic:1027(quit) DEBUG: Session: {'theme': 'day', 'mode': 'python', 'paths': ['/Users/konoha/Desktop/VIS142/project1/project1.py'], 'envars': [], 'minify': False, 'microbit_runtime': ''}
2018-10-30 23:33:11,580 - mu.logic:1028(quit) DEBUG: Saving session to: /Users/konoha/Library/Application Support/mu/session.json
2018-10-30 23:33:11,580 - mu.logic:1030(quit) INFO: Quitting.

```

96konoha, 2018-10-31:
The crash happened on 2018-10-30 23:20, the file is saved as an empty file then.....???!

96konoha, 2018-10-31:
Search for "2018-10-30 23:20:20,953" in the log

tjguk, 2018-10-31:
The specific answer here is that the code in modes/python3.py:run_script needs to call save_and_encode rather than write_and_flush. And presumably the equivalent code in other modes as well; I'll check.

On a slightly wider note, I propose to add a fallback save somewhere which will do an emergency write-out using latin-1 (which will encode anything) if an exception is caught by the primary code. This will give us something of a recovery position if this kind of situation happens. We can also pop up a message indicating what's happened

tjguk, 2018-11-25:
I can't reproduce this bug in the current version of Mu on Win10. I have no problem in making the switch from write_and_flush to save_and_encode. But I can't seem to manage a failing test. I've tried the code from this issue and from #705 and it saves correctly.

Can anyone reproduce using the current HEAD?

I'll try to backtrack to the 1.0.1 release to see if I can reproduce from that, in which case I would assume we've inadvertently fixed the problem already.

tjguk, 2018-11-25:
Checked out the 1.0.1 tag and the test still succeeds.

chumbers, 2019-01-03:
I started using Mu 1.0.1 today and reproduced this.  I copied and pasted Chinese characters from https://en.wikipedia.org/wiki/Chinese_language into a print statement: 国语/國語
Mu disappeared when I ran the script.
When re-launching Mu, my hello.py script was empty.
The crash did not reproduce on the second attempt.

The entirety of my script was:
```
print("Hello from Mu!")
print('国语/國語')
```

The relevant portion of the log appears to be:

```
2019-01-03 11:09:05,915 - root:95(excepthook) ERROR: Unrecoverable error
Traceback (most recent call last):
  File "C:\Users\chumbers\AppData\Local\Mu\pkgs\mu\modes\python3.py", line 162, in run_toggle
    self.run_script()
  File "C:\Users\chumbers\AppData\Local\Mu\pkgs\mu\modes\python3.py", line 189, in run_script
    write_and_flush(f, tab.text())
  File "C:\Users\chumbers\AppData\Local\Mu\pkgs\mu\logic.py", line 152, in write_and_flush
    fileobj.write(content)
  File "encodings\cp1252.py", line 19, in encode
UnicodeEncodeError: 'charmap' codec can't encode characters in position 32-33: character maps to <undefined>
2019-01-03 11:09:15,108 - root:112(run) INFO: 
```
I have python 3.7 (32-bit) installed.
The entire Log is attached.
[log.txt](https://github.com/mu-editor/mu/files/2724764/log.txt)

carlosperate, 2019-01-03:
The fix hasn't been released yet, could you try the latest installed from this list and see if it works in that version?
http://mu-builds.s3-website.eu-west-2.amazonaws.com/?prefix=windows/

chumbers, 2019-01-03:
It works, but I don't think it is a valid test.  It appears that the crash is intermittent, or that I don't know the right steps to guarantee the crash.  I couldn't reproduce the crash with 1.0.1 on another Windows 10 machine.

I did the following to test with Master:
1. Uninstalled Mu 1.0.1
2. Installed  mu_2018-12-21_19_31_master_3157ce3_64bit.exe
3. I launched Mu and it came up with my previous file already open.
4. I clicked Run and the script executed properly.

I then uninstalled the build from Master and re-installed 1.0.1. No crash.

jamesabela, 2019-01-04:
The same has happened to me on my mac. Unfortunately, because it wiped the code I can't share it.  However it might have something to do with Running code without saving it first. I am on version 1.0.0

ntoll, 2019-01-04:
Hi folks,

This issue will be fixed in the upcoming 1.0.2 release.

dybber, 2021-02-06:
> Hi folks,
> 
> This issue will be fixed in the upcoming 1.0.2 release.

@ntoll: did you fix this already? We already have a 1.0.2 release, and I guess you included a fix, or what is the status here? If you remember... at least we haven't heard any new reports about it for two years.

ntoll, 2021-02-14:
Hi @dybber

Hmm... I've just taken a look at the code. The TL;DR is that I am an idiot and @carlosperate is not. :-)

Here's the more in-depth situation:

* @carlosperate fixed this in the #208 pull request.
* However, this introduced a bug in CircuitPython mode - every time there is a write, the board restarts. Since there are two writes with the `AtomicFile` class (the first one being a temp backup), then the CircuitPython boards would re-start before the second "real" write to the device.
* In an attempt to fix the CircuitPython issue, I deleted/reverted the atomic write code that Carlos introduced (an over-eager solution on my part).
* We should do two things:
  * Re-introduce Carlos's code from #208.
  * Fix that code so it doesn't run when in CircuitPython mode. I suggest a simple flag called `atomic` which defaults to `True` when we save a file. In CircuitPython mode, we can switch the flag to `False` and "normal" Python writes to the device happen (surrounded by a `try`/`except` just in case).

Sound like a plan..?

carlosperate, 2021-02-14:
It's been a long time since I looked at the atomic write, but if I remember correctly we could also change the temp file to be in a different directory (like the OS temp dir), so that the last step (the file move performed kind of atomically) can still be done the same, just not from within the CIRCUITPYTHON drive.
For other modes it can still be useful to have the temp file next to the original, as after a Mu crash the user can still easily retrieve it.

ntoll, 2021-02-14:
Exactly. :+1:

ntoll, 2021-07-10:
I believe this problem is guarded against given the current state of the code that handles file saves (see `save_tab_to_file` in `logic.py`). Would be good if someone could take a _quick_ look and confirm. If so we can close this issue.

rp10007, 2022-01-04:
Perhaps I'm wrongheaded, but from a circuitpython workflow point of view having a configurable place to put copies of stuff written to flash would be delightful. I have a class with multiple groups working on similar bots, and sharing/versioning code that lives only on the flash drives is a pain. If a copy of the code lived somewhere on the host machine (without the back-and-forth of repeated "Save As") some of the interaction among students would be much easier. So I'm cheering on whoever is working on this infrastructure.