mu-editor/mu #697: Unable to import

LargeKnome, 2018-11-01:
[Log.txt](https://github.com/mu-editor/mu/files/2538550/Log.txt)
Here is my full log, as requested.

carlosperate, 2018-11-12:
Hi @LargeKnome,

Thank you for the report, but without a bit more info we don't really know what to look for.

Could you explain step by step what you were trying to do, what did you expect to happen, and what did Mu do?

tjguk, 2018-11-13:
@carlosperate for info this was the outcome from a discussion over on Gitter starting around here: https://gitter.im/mu-editor/general?at=5bdae32ea9c55b25748d40c1

It's not an awful lot clearer there as it's difficult to get a clear picture on an online to-and-fro. But it does seem to be somehow tied to Mu having been installed in an unusual location to start with and/or the files being picked up from somewhere unusual and then moved over.

I suspect (in fact I half remember from when I was doing the load/save testing) that we have odd corner cases where two files have the same name but in different directories. But I haven't put any time into trying to reproduce that.

carlosperate, 2018-11-13:
Thank you @tjguk.
This is definitely https://github.com/mu-editor/mu/issues/662, we need to fix including `mu_code` into the path for Windows. It works in debug, but not in run.

carlosperate, 2020-09-29:
As #662 has been fixed this should be resolved, so I'll close this issue.
If anybody is still experiencing this problem please reopen this issue or #662.
Thanks for the report @LargeKnome!

closed 2020-09-29 by carlosperate