mu-editor/mu #698: Distribute Mu as a snap

bennuttall, 2018-11-06:
I'm working with @flexiondotorg and @waveform80 on publishing a [snap](https://en.wikipedia.org/wiki/Snappy_(package_manager)) for Mu, which will make it easy to install Mu across Linux distributions and potentially Windows and Mac in future. When a snap is published in the [snapcraft store](https://snapcraft.io), Mu will be installable with `snap install mu-editor` and will bundle all dependencies, including Python, QT and all python libs.

We will introduce a `snapcraft.yml` into the root of the project. This will contain the spec for how the snap is built.

carlosperate, 2018-11-06:
I've never done snap packaging before, but one thing to keep in mind would be the sandboxing. I assume access to the file system to open files is fine, but Mu saves config data in the user locations indicated by the `appdirs` python module, does that still work correctly?

Also, Mu launches the debugger as a separate process and connects to it via localhost, so that's something to check as well if it works. 

Apart from that, the other thing I'd keep an eye on is having access to the serial for the micro:bit and adafruit modes.

bennuttall, 2018-11-06:
Yep - the debugger works

This comes up when launching:

```
ben@bionic:~/Projects/mu$ snap run mu-editor
Logging to /home/ben/snap/mu-editor/common/.cache/mu/log/mu.log
```

Also this:

```
sh: 0: getcwd() failed: No such file or directory
```

Yeah, thinking about serial and the ability to flash the device.

bennuttall, 2018-11-06:
Progress! A partially functional Mu v1.0.0 is now released in the snap store!

Linux users with snap support, try this:

```bash
snap install mu-editor --edge
```

Launch from your application drawer/menu or form the command line with `snap run mu-editor`.

We need to figure out USB permissions and whatnot, but the Python 3 and pgzero stuff works!

bennuttall, 2018-11-06:
One thing to consider is that there's no clear way (I can think of) to install additional libraries. It's not using the system Python.

carlosperate, 2018-11-06:
Awesome news!

About installing packages, when https://github.com/mu-editor/mu/issues/675 is implemented it should cover this scenario by default, as I assume snap packs its own Python interpreter, and the built-in solution would install the packages in that environment.

In the meantime, as a workaround, packages can be installed via REPL:

[image]

bennuttall, 2018-11-06:
I've fast-forwarded to the 1.0.1 tag and released this version too.

bennuttall, 2018-11-06:
Also I've moved this work into the `snap` branch here. Thanks to @waveform80 and @flexiondotorg for the work getting us this far!

I'm out of sync with master due to having to rip out some stuff from `setup.py`. I'm sure we can piece it back together at some point though.

flexiondotorg, 2018-11-07:
@bennuttall Please take a look at https://github.com/flexiondotorg/mu/commit/a65b58e67b8731e58c239832b206b6634cac3a76

It adds `pip3` to the snap and also exposes the command `mu-editor.pip`. With the mu-editor snap installed you can now run something like `mu-editor.pip install requests` which will install `requests` to the user writable common data area inside the mu-editor snap :smiley: 

As for accessing devices over USB, the `raw-usb` interface may help here but will (currently) require manual connection using `snap connect mu-editor:raw-usb`. With `raw-usb` interface connected does this improve device access?

waveform80, 2018-11-07:
@ZanderBrown If the block is the PyQt *version* then it's worth noting that the snap uses PyQt 5.5 (the requirements listed in the setup.py are much later than they actually need to be). In other words, the system PyQt5 shipped with Ubuntu Xenial or Raspbian Stretch are both sufficient to run mu

carlosperate, 2018-11-07:
Just a note on PyQt versions, there were quite a few issues that turned out to be bugs in specific PyQt versions (a lot of them were platform specific, so not all will apply to Linux), so there might be some regressions if using an older version. I don't recall any show stoppers, but something to keep an eye if something misbehaves.

Also PyQtChart was only available since PyQt5.7 (before it was a commercial module). Does that mean the snap can't have the plotter?

waveform80, 2018-11-07:
> Also PyQtChart was only available since PyQt5.7 (before it was a commercial module). Does that mean the snap can't have the plotter?

Yes, that's correct - the xenial-base libs used in the snap can't support the plotter. The bionic-base libs (when they become available) *might* do but I haven't checked whether they've got all the necessary bits either.

waveform80, 2018-11-08:
Well, I'm reliably informed core18 (the bionic base) has been available for some time now! Might be worth throwing that in the snap and seeing if you can get PyQtChart working via that (core18 should have Qt 5.10 IIRC)

carlosperate, 2018-11-08:
That's great news!
I assume that a snap with core18 would also work on 16.04?

flexiondotorg, 2018-11-08:
Yes, core18 provides the base runtime snaps sit atop. Snaps using core18 will still run on all the distros where snapd is available 🙂

flexiondotorg, 2018-11-15:
If you add the `pulseaudio` interfaces to the `plugs:` in the `snapcraft.yaml` then audio should work.

waveform80, 2018-11-16:
Add libdb to stage-packages? :)