mu-editor/mu #699: Saving Paths - windows Version

damianburrin, 2018-11-07:
@ntoll Are the saving paths hard-coded?  I seem to be having problems, students don't have access to drive C on our network but when the students try to change the file save or open path it gives permission denied errors as if its still trying to read/write from the local mu folder/

Thanks

If you are reporting a bug, we would like to know:

* What you were trying to do,
* What steps you took to make this happen,
* What you expected to happen,
* What actually happened,
* Why this difference is problematic (it may not be a bug!),
* Technical details like the version of Mu you're using, your OS version and
  other aspects of the context in which Mu was running. 

Please remember to attach a **copy of the full log files for Mu**. You can get
the logs by clicking on the cog icon in the bottom right of the editor window.
Click on the logs and use CTRL-A to select all, then CTRL-C to copy and CTRL-V
to paste the contents into the issue.

Thank you for contributing to Mu! :-)

tjguk, 2018-11-07:
@damianburrin This is a bit fiddly. For the "mu_code" location which is where user files are stored, Mu uses Python's os.path.expanduser functionality. On Windows this will resolve to (slightly simplified and in the following order):

%HOME%
%USERPROFILE%
%HOMEDRIVE%\%HOMEPATH%

It's unusual to have %HOME% set so the likelihood is %USERPROFILE%. By default on recent Windows that will be c:\users\<username> but it's possible for it to have been redirected by Windows admins to point to, eg, a network share etc. A more likely redirect is HOMEDRIVE/PATH or HOMESHARE (which is more recent and which Python doesn't check).

This slightly extended explanation is really to say: Mu is almost certainly trying to read/write from c:\users\<username>\mu_code. If you were drop the log files from a user session into this issue we could see exactly where it was trying.

tjguk, 2018-11-07:
@damianburrin I'm fairly familiar with schools having locked-down systems, but preventing access to the user's own home directory seems as little excessive. And, if that's the case, we can't be the only application having problems. Have I misunderstood the situation?

tjguk, 2018-11-07:
@damianburrin Sorry, just occurred to me -- assuming everything is as it appears, a fairly quick workaround might be to set the HOME env var for each users to something acceptable (eg an accessible network share).

damianburrin, 2018-11-07:
@tjuk At my school all direct access to the C drive is blocked.  The users home directory is a UNC path to their user folder on the server.

I'll speak to the techy and see what the home env variable is set too

tjguk, 2018-11-07:
Thanks; could you also check the other env vars I mentioned: USERPROFILE, HOMEDRIVE/HOMEPATH?

damianburrin, 2018-11-07:
@tjuk Will do - I've also asked if he can send me a copy of the error log to upload

tjguk, 2018-11-07:
Thanks -- for the avoidance of doubt, the error log I'm looking for is the one from Mu itself (the cog in the bottom-right hand corner). I'd expect you to be able to pull that off yourself.

bennuttall, 2018-11-08:
Note you can set env vars within Mu, assuming that works ok on Windows?

[image]

tjguk, 2018-11-08:
Thanks for the reminder, Ben. That's more of a runtime feature though: this env var needs to be set before you run Mu itself.

tjguk, 2018-11-09:
I'd be interested to hear back on this, @damianburrin. I suspect we may need to work harder on Windows to find an appropriate place for the mu_code folder in the face of redirected home shares.

damianburrin, 2018-11-09:
Hi @tjguk   I've not heard anything back from our tech guys.  You know what it;s like in schools.  They have put in a work round which does seem to solve the issue.  They have mapped the users redirected mydocs to a drive letter and then this seems to let you select and save.  Obviously we can't import libraries as they need to be in the Mu-code folder but at least we can save now :-)  Normally students don't have a fixed drive letter, but when they choose my docs this re-directs to their network share.

tjguk, 2018-11-10:
Thanks for the update, @damianburrin. I would rather like to understand exactly what setup they have in place (because it's likely to be in place in other schools). But I recognise that this is hardly going to be anyone's priority, especially since they have a workaround. FWIW I can't think of any reason why assigning a drive letter should grant permissions where a UNC or a redirected shell folder doesn't. That said, there's always something new to learn...

tjguk, 2018-11-10:
I think I'll close this one for now as the immediate need has been satisfied by a workaround. But I'll raise a separate issue on the general question of where the mu_code folder should sit (especially on Windows)

tjguk, 2018-11-10:
To my embarrassment I completely missed the fact that we already allow an override in the settings.json file. Your admins (or you) would still have to get this file into the right place for each student, but that might not be such a problem.

And we even have a tutorial about it. Could you try this guide, please, to see if it addresses the issue:

https://codewith.mu/en/tutorials/1.0/configuration

tjguk, 2018-11-12:
I've had a quick discussion with @ntoll and I'm going to close this issue and create a new one which proposes that we detect when Mu starts for the first that our preferred "home" directory is not writable. At that point we'd pop up a simple dialog box asking for a valid alternative. This will then be written to a settings.json file and used in the future.

closed 2018-11-12 by tjguk

damianburrin, 2018-11-12:
Awesom.  Thank you