mu-editor/mu #703: Update settings.json for mu_code when default is not writeable

tjguk, 2018-11-12:
(Following on from #699)

Mu places its mu_code folder in os.expandvar{"~") by default. On Windows this is likely to resolve to %USERPROFILE% (usually c:\users\<username>). In some school or other corporate network settings, users have a network-shared home directory and the %USERPROFILE% folder is either not used or is not in fact writable by the user.

Users can already add the workspace option to settings.json (although I don't think that resolves env vars which might be an interesting addition). And this is an option which school admins could take up, including pushing out settings.json files to individual computers. [Note to self: find out where settings.json is discovered and whether that could also be a network path].

We propose here to detect the fact that our chosen location for mu_code is not user-writeable and offer the user a simple input dialog in which they enter a valid mu_code path. This is then written to settings.json on the user's behalf and will be used for all future instantiations of the editor. 

Otherwise, the default will be written to settings.json which has the useful side-effect that the absence of a workspace option in settings.json is the trigger for this check to take place.

carlosperate, 2018-11-12:
> ... and offer the user a simple input dialog in which they enter a valid mu_code path. This is then written to settings.json on the user's behalf and will be used for all future instantiations of the editor.

It's been discussed a few times to expose some of these options into the settings window. Perhaps this might be a good opportunity to do that?

tjguk, 2018-11-12:
@carlosperate I think I see these things as slightly orthogonal. Ultimately everything ends up in settings.json or session.json. How it gets there could include: direct user intervention (including some kind of admin "push"); a GUI in the settings window; a detect-and-update mechanism of the kind I'm suggesting here.

carlosperate, 2018-11-12:
Yes, but let's say we already had the option in the settings, in that case I would say the detect-and-update mechanism doesn't need an additional window with a path field, and instead it could just send the user to the the settings window (you still need some type of alert window to inform the user of what's about to happen and why, but not a second one with just the path field).

However, now that I'm writing it down I realise perhaps you were picturing a single popup with the message and the field in one window, in which case what I am suggesting doesn't really save us any additional windows, and would make a lot more sense. Not sure why I understood this as being a 2-steps process.

However, I would also argue that it's quite possible a lot of users (specially disinterested ones) might just pick a random location and then forget where it was. Without having the option in the settings window by the time this is shipped it would be quite difficult to change this setting back to something more sensible.

johnsturgess, 2018-11-15:
What is a working example for settings.json on Windows?

I tried using a variable (until i read the above comment) and a mapped drive path but I am still receiving an error saying that the settings file couldn't be parsed.

```
{
  "workspace": "N:\Documents\MuCode"
} 
```

tjguk, 2018-11-26:
(To remind myself) I've started work on a write-up of the current situation over at https://github.com/tjguk/mu/tree/issue703-settings-json