mu-editor/mu #705: 'charmap' codec can't encode character '\U0001f60a'

martinohanlon, 2018-11-14:
I was testing a program for a student on one of our courses and Mu unexpected crashed.

Here is the log. I can only assume it is due to the "smiley" I hadnt noticed in the print.

```
2018-11-14 07:55:55,431 - mu.logic:1138(autosave) INFO: Autosave detected and saved changes in C:\Users\Martin O'Hanlon-LT\mu_code\tes7.py.
2018-11-14 07:55:57,171 - mu.modes.python3:207(stop_script) DEBUG: Stopping script.
2018-11-14 07:55:57,610 - mu.modes.python3:187(run_script) INFO: Saving script to: C:\Users\Martin O'Hanlon-LT\mu_code\tes7.py
2018-11-14 07:55:57,610 - mu.modes.python3:188(run_script) DEBUG: my_name = "Marlene"

print("Hello, I'm " + my_name + ".  I’m going to help you work out each person’s bill today")

menu = []

count = 0

friend_number = input("how many people are in your group today? ")

friend_number = int(friend_number)

phrase = input("is everyone sharing the bill equally? ")

if phrase == "yes" or phrase == "sure" or phrase == "okay":

  print("Okay. Great!")

how_many = input("how many items did you order? ")

how_many = int(how_many)

for item_number in range(how_many):

  item = input("what‘s  dish " + str(item_number + 1) + "? ")

  menu.append(item)

  total = 0

  total = float(total)

for item in menu:

  cost = input("what’s the cost of " + item + "? > ")

  cost = float(cost)

  total = total + float(cost)

result = total / friend_number

print("Okay, the total cost of the items is $" + str(total))

print("each person has to pay $" + str(result) + " without a tip")

def tip():

  tip_calcu = input("What percentage tip do you want to leave? > ")

  tip_calcu = float(tip_calcu)

  tip = total * (tip_calcu / 100)

  total2 = total + tip

  result2 = total2 / friend_number

  print("A" + str(tip_calcu) + "% tip would be $" + str(tip))

tip()  

print("So, with a tip, everyone has to pay $" + str(result2))

print("Hope you all enjoyed the meal!  😊")
2018-11-14 07:55:57,610 - root:95(excepthook) ERROR: Unrecoverable error
Traceback (most recent call last):
  File "C:\Users\Martin O'Hanlon-LT\AppData\Local\Mu\pkgs\mu\modes\python3.py", line 162, in run_toggle
    self.run_script()
  File "C:\Users\Martin O'Hanlon-LT\AppData\Local\Mu\pkgs\mu\modes\python3.py", line 189, in run_script
    write_and_flush(f, tab.text())
  File "C:\Users\Martin O'Hanlon-LT\AppData\Local\Mu\pkgs\mu\logic.py", line 152, in write_and_flush
    fileobj.write(content)
  File "encodings\cp1252.py", line 19, in encode
UnicodeEncodeError: 'charmap' codec can't encode character '\U0001f60a' in position 1293: character maps to <undefined>
2018-11-14 07:56:02,790 - root:112(run) INFO: 

-----------------
```

Interestingly I couldn't get it to replicate. I can only assume I have lost the original encoding through cut and paste.

tjguk, 2018-11-14:
Thanks @martinohanlon (and sorry for the disruption to your student). This is a duplicate of #696 so I'll close it here and try to get a PR ready over there Real Soon Now (tm).

closed 2018-11-14 by tjguk

martinohanlon, 2018-11-14:
Thanks Tim. No problem.