mu-editor/mu #707: stdout is truncated at 512 characters

bennuttall, 2018-11-14:
I've found that and stdout is limited to 512 characters.

```python
print('|'*511 + 'x')
```

works as expected. I see 511 pipes followed by `x`:

```
|||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||x
```

However:

```python
print('|'*512 + 'x')
```

I only see 512 pipes, no `x`:

```
||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||
```

This seems to be the cause of the pgzero issue I opened earlier: #706.

tjguk, 2018-11-14:
Thanks, Ben. This seems to be a popular issue today -- and there's a PR #685 currently waiting to be tested to fix this

ntoll, 2018-12-13:
Fixed by #723

closed 2018-12-13 by ntoll