mu-editor/mu #709: ModuleNotfoundError

stemman, 2018-11-20:
If you are reporting a bug, we would like to know:

* What you were trying to do,
* What steps you took to make this happen,
* What you expected to happen,
* What actually happened,
* Why this difference is problematic (it may not be a bug!),
* Technical details like the version of Mu you're using, your OS version and
  other aspects of the context in which Mu was running. 

Please remember to attach a **copy of the full log files for Mu**. You can get
the logs by clicking on the cog icon in the bottom right of the editor window.
Click on the logs and use CTRL-A to select all, then CTRL-C to copy and CTRL-V
to paste the contents into the issue.

Thank you for contributing to Mu! :-)

stemman, 2018-11-20:
Hello sorry for this silly question but how do i tell mu-editor to use my existing packages in my windows 3.6? I have pandas package install in C:\Python36\Lib\site-packages, but don't know how to tell mu_editor to look for the package in that directory.

carlosperate, 2018-11-20:
Hi @stemman,

Mu (if installed from the Windows or macOS installer) contains its own copy of Python and packs with it a number of useful packages.
At the moment is not possible to point Mu to a package installed in your system Python, but there are plans to introduce a way to install your own packages within Mu. More info can be found in https://github.com/mu-editor/mu/issues/675.

Alternatively, if you install Mu using pip ([instructions](https://codewith.mu/en/howto/install_with_python)) then it will have access to all the packages available in that pip environment. However, it is not recommended to install applications like Mu in your system Python installation, as packages could conflict with other installed applications/packages/scripts, and it is always  recommended to use a virtual environment.
Thankfully, if you create a virtual environment for Mu, you can always install any packages you need inside that virtualenv, including pandas.

If you really need to, you could use this workaround to install pandas inside a Mu installation: https://github.com/mu-editor/mu/issues/698#issuecomment-436428145
However, I would consider this an advance workaround, if something goes wrong it can be hard to debug, so only do it if you feel familiar enough with Python to fix things if something breaks (or I guess if you are happy to uninstall and reinstall Mu if something goes wrong).

closed 2020-09-27 by dybber