mu-editor/mu #711: Changing mode via alert box when a board is plugged keeps previous panes open

carlosperate, 2018-11-21:
To replicate:

* Set the pygamezero mode
* Click "Run" and leave the window open
* Plug in a micro:bit or circuitplayground
* An alert box asks if you want to change modes, click "yes"
* Mode is changed but the panes from the previous mode are still open

This is because the feature introduced in https://github.com/mu-editor/mu/pull/452 delegates change mode logic to the `Editor.change_mode` method, and since the "Mode" button is deactivated when running a game, this method never had to worry about closing that panel.

I think a refactor to the panes was discussed previously (to be able to keep some features between runs or opening/closing panes) I think at that point it would be good to add a general "open/close all panes" method to each mode, so that the individual modes are responsible to keep track of this feature. Otherwise changes to the panes within a mode would have to result in changes to the `Editor.change_mode` method, which might not be obvious unless you are aware of it: https://github.com/mu-editor/mu/blob/5a7e3dfc91ad34406dc11f662510a70c9f628c38/mu/logic.py#L1077-L1090

ntoll, 2018-12-19:
The fix I just landed in master is *temporary* to mitigate the immediate. I agree that each mode should have a sort of "destructor" which can be called to clean itself up while changing modes. We'll do that in 1.1 (the focus of 1.0.2 is bug fixes). See #530 to track this work.

closed 2018-12-19 by ntoll

carlosperate, 2018-12-19:
Sounds good, thank you Nicholas!

commit e23ba95c (2018-12-19): Fixes #711. Ensure no mode change prompt if current mode is running code.