mu-editor/mu #714: Feature - where is plotter?

rhubarbdog, 2018-11-29:
The screen shot at codewith.mu shows a copy of mu with a plotter button.
Is there a version with a plotter?
Is there any documetation and examples of graphing data?

wayoda, 2018-11-29:
There is a [Tutorial for the Plotter](https://codewith.mu/en/tutorials/1.0/plotter) on the[Code with Mu Tutorials](https://codewith.mu/en/tutorials/) page.

wayoda, 2018-11-29:
Sorry, I never installed it on an RPi. Can't help on this one.

ukBaz, 2018-11-29:
There are issues with the plotter on an RPi. See:
https://github.com/mu-editor/mu/issues/689#issuecomment-433419161

carlosperate, 2018-11-30:
We are going through the process to create PyQtChart (the library that provides the plotted) deb packages, which can then be submitted for inclusion in Debian and/or Raspbian.

Issue  https://github.com/mu-editor/mu/issues/441 is the current thread discussing this work, so I'll close this one as a duplicate. Thanks @wayoda  and @ukBaz for all your help as well.

closed 2018-11-30 by carlosperate