mu-editor/mu #717: problems with files button in Mu when file is 20kb

rhubarbdog, 2018-12-06:
I was trying to move a large file from my microbit to my pc using Mu
I plugged in my microbit, started Mu and clicked the 'files' button

I hoped to see my file in the left hand lower pane, it was empty and the Mu displayed a dialogue
"""
There was a problem getting the list of files on the micro:bit. Please check Mu's logs for technical information. Alternatively, try unplugging/plugging-in your micro:bit and/or restarting Mu.
"""

 i clicked files to close the panes and clicked again to try and it's there???
so i try to copy it over with drag and drop and get the following dialogue
"""
There was a problem getting '20k.txt' from the micro:bit. Please check Mu's logs for more information.
"""

I'm using Mu version 1.0.0
on Linux Mint 18.2 which is a clone of ubuntu 16.04
this may be a duplicate of <a href="https://github.com/ntoll/microfs/issues/12">@ntol/microfs/#12</a>

```
2018-12-06 05:05:35,443 - root:112(run) INFO: 

-----------------

Starting Mu 1.0.0
2018-12-06 05:05:35,443 - root:113(run) INFO: uname_result(system='Linux', node='samsung01', release='4.13.0-32-generic', version='#35~16.04.1-Ubuntu SMP Thu Jan 25 10:13:43 UTC 2018', machine='x86_64', processor='x86_64')
2018-12-06 05:05:35,443 - root:114(run) INFO: Python path: ['/home/usrlocal/local/bin/mu-master', '/usr/lib/python35.zip', '/usr/lib/python3.5', '/usr/lib/python3.5/plat-x86_64-linux-gnu', '/usr/lib/python3.5/lib-dynload', '/home/phil/.local/lib/python3.5/site-packages', '/usr/local/lib/python3.5/dist-packages', '/usr/lib/python3/dist-packages', '/usr/local/lib/python3.5/dist-packages/IPython/extensions']
2018-12-06 05:05:35,571 - mu.logic:540(__init__) INFO: Setting up editor.
2018-12-06 05:05:35,571 - mu.logic:558(__init__) INFO: Settings path: /home/phil/.local/share/mu/settings.json
2018-12-06 05:05:35,571 - mu.logic:559(__init__) INFO: Session path: /home/phil/.local/share/mu/session.json
2018-12-06 05:05:35,571 - mu.logic:560(__init__) INFO: Log directory: /home/phil/.cache/mu/log
2018-12-06 05:05:35,571 - mu.logic:561(__init__) INFO: Data directory: /home/phil/.local/share/mu
2018-12-06 05:05:35,612 - mu.logic:574(setup) INFO: Available modes: python, debugger, adafruit, microbit, pygamezero
2018-12-06 05:05:35,794 - mu.logic:1110(change_mode) INFO: Workspace directory: /home/phil/mu_code
2018-12-06 05:05:35,795 - mu.logic:623(restore_session) INFO: Restoring session from: /home/phil/.local/share/mu/session.json
2018-12-06 05:05:35,795 - mu.logic:624(restore_session) DEBUG: {'theme': 'day', 'mode': 'microbit', 'minify': True, 'paths': [], 'microbit_runtime': '', 'envars': []}
2018-12-06 05:05:35,795 - mu.logic:645(restore_session) INFO: Loaded files.
2018-12-06 05:05:35,795 - mu.logic:649(restore_session) INFO: User defined environment variables: []
2018-12-06 05:05:35,795 - mu.logic:653(restore_session) INFO: Minify scripts on micro:bit? True
2018-12-06 05:05:35,998 - mu.logic:671(restore_session) INFO: Starting with blank file.
2018-12-06 05:05:36,029 - mu.logic:1110(change_mode) INFO: Workspace directory: /home/phil/mu_code
2018-12-06 05:05:41,445 - mu.logic:1037(show_admin) INFO: Showing logs from /home/phil/.cache/mu/log/mu.log
2018-12-06 05:05:56,512 - mu.modes.base:217(find_device) INFO: Found device on port: ttyACM0
2018-12-06 05:05:56,512 - mu.modes.base:218(find_device) INFO: Serial number: 9900023431634e45007940010000003800000000cca828bd
2018-12-06 05:05:56,523 - mu.modes.microbit:597(toggle_files) INFO: Toggle filesystem on.
2018-12-06 05:05:57,244 - mu.modes.microbit:122(ls) ERROR: unexpected indent (<unknown>, line 1)
Traceback (most recent call last):
  File "/home/usrlocal/local/bin/mu-master/mu/modes/microbit.py", line 119, in ls
    result = tuple(microfs.ls())
  File "/home/usrlocal/local/bin/mu-master/mu/contrib/microfs.py", line 184, in ls
    return ast.literal_eval(out.decode('utf-8'))
  File "/usr/lib/python3.5/ast.py", line 46, in literal_eval
    node_or_string = parse(node_or_string, mode='eval')
  File "/usr/lib/python3.5/ast.py", line 35, in parse
    return compile(source, filename, mode, PyCF_ONLY_AST)
  File "<unknown>", line 1
    CTRL-B to exit
    ^
IndentationError: unexpected indent
2018-12-06 05:05:57,250 - mu.interface.main:701(show_message) DEBUG: There was a problem getting the list of files on the micro:bit. Please check Mu's logs for technical information. Alternatively, try unplugging/plugging-in your micro:bit and/or restarting Mu.
2018-12-06 05:05:57,250 - mu.interface.main:702(show_message) DEBUG: None
2018-12-06 05:13:32,694 - mu.modes.microbit:601(toggle_files) INFO: Toggle filesystem off.
2018-12-06 05:13:34,018 - mu.modes.base:217(find_device) INFO: Found device on port: ttyACM0
2018-12-06 05:13:34,018 - mu.modes.base:218(find_device) INFO: Serial number: 9900023431634e45007940010000003800000000cca828bd
2018-12-06 05:13:34,024 - mu.modes.microbit:597(toggle_files) INFO: Toggle filesystem on.
2018-12-06 05:13:43,442 - mu.logic:1037(show_admin) INFO: Showing logs from /home/phil/.cache/mu/log/mu.log
2018-12-06 05:15:22,608 - mu.interface.panes:388(dropEvent) INFO: Getting '20k.txt' from micro:bit. Copying to '/home/phil/mu_code/20k.txt'.
2018-12-06 05:15:24,001 - mu.modes.microbit:135(get) ERROR: not enough values to unpack (expected 2, got 1)
2018-12-06 05:15:24,004 - mu.interface.main:701(show_message) DEBUG: There was a problem getting '20k.txt' from the micro:bit. Please check Mu's logs for more information.
2018-12-06 05:15:24,004 - mu.interface.main:702(show_message) DEBUG: None
2018-12-06 05:17:17,154 - mu.logic:1037(show_admin) INFO: Showing logs from /home/phil/.cache/mu/log/mu.log
```

wayoda, 2018-12-06:
My guess is that mu treats the file as a python module when copying. It looks like python modules should not have a name that starts with a number. Could you to rename the file from `20k.txt`to `k20.txt` and try again?

I found a similar problem on [StackOverflow-In python, how to import filename starts with a number
](https://stackoverflow.com/questions/9090079/in-python-how-to-import-filename-starts-with-a-number).

rhubarbdog, 2018-12-06:
renamed the file to `k20k.txt`
"""
There was a problem getting 'k20k.txt' from the micro:bit. Please check Mu's logs for more information.
"""
```
2018-12-06 10:41:42,354 - root:112(run) INFO: 

-----------------

Starting Mu 1.0.0
2018-12-06 10:41:42,354 - root:113(run) INFO: uname_result(system='Linux', node='samsung01', release='4.13.0-32-generic', version='#35~16.04.1-Ubuntu SMP Thu Jan 25 10:13:43 UTC 2018', machine='x86_64', processor='x86_64')
2018-12-06 10:41:42,354 - root:114(run) INFO: Python path: ['/home/usrlocal/local/bin/mu-master', '/usr/lib/python35.zip', '/usr/lib/python3.5', '/usr/lib/python3.5/plat-x86_64-linux-gnu', '/usr/lib/python3.5/lib-dynload', '/home/phil/.local/lib/python3.5/site-packages', '/usr/local/lib/python3.5/dist-packages', '/usr/lib/python3/dist-packages', '/usr/local/lib/python3.5/dist-packages/IPython/extensions']
2018-12-06 10:41:43,086 - mu.logic:540(__init__) INFO: Setting up editor.
2018-12-06 10:41:43,097 - mu.logic:558(__init__) INFO: Settings path: /home/phil/.local/share/mu/settings.json
2018-12-06 10:41:43,097 - mu.logic:559(__init__) INFO: Session path: /home/phil/.local/share/mu/session.json
2018-12-06 10:41:43,097 - mu.logic:560(__init__) INFO: Log directory: /home/phil/.cache/mu/log
2018-12-06 10:41:43,098 - mu.logic:561(__init__) INFO: Data directory: /home/phil/.local/share/mu
2018-12-06 10:41:43,138 - mu.logic:574(setup) INFO: Available modes: python, pygamezero, microbit, debugger, adafruit
2018-12-06 10:41:43,834 - mu.logic:1110(change_mode) INFO: Workspace directory: /home/phil/mu_code
2018-12-06 10:41:43,848 - mu.logic:623(restore_session) INFO: Restoring session from: /home/phil/.local/share/mu/session.json
2018-12-06 10:41:43,848 - mu.logic:624(restore_session) DEBUG: {'microbit_runtime': '', 'theme': 'day', 'minify': True, 'envars': [], 'mode': 'microbit', 'paths': []}
2018-12-06 10:41:43,849 - mu.logic:645(restore_session) INFO: Loaded files.
2018-12-06 10:41:43,849 - mu.logic:649(restore_session) INFO: User defined environment variables: []
2018-12-06 10:41:43,849 - mu.logic:653(restore_session) INFO: Minify scripts on micro:bit? True
2018-12-06 10:41:44,182 - mu.logic:671(restore_session) INFO: Starting with blank file.
2018-12-06 10:41:44,209 - mu.logic:1110(change_mode) INFO: Workspace directory: /home/phil/mu_code
2018-12-06 10:41:58,770 - mu.interface.main:235(get_load_path) DEBUG: Getting load path: /home/phil/mu_code/pongo.py
2018-12-06 10:41:58,771 - mu.logic:706(_load) INFO: Loading script from: /home/phil/mu_code/pongo.py
2018-12-06 10:41:58,771 - mu.logic:271(read_and_decode) DEBUG: Trying to decode with utf-8
2018-12-06 10:41:58,771 - mu.logic:274(read_and_decode) INFO: Decoded with utf-8
2018-12-06 10:41:58,772 - mu.logic:287(read_and_decode) DEBUG: Detected newline '\n'
2018-12-06 10:41:58,772 - mu.logic:780(_load) DEBUG: from microbit import display
import gamezip
import random

BAT_LEN = 3
bat_x =  2
BAT_Y = 7
BAT_COL = (30, 0, 0)

ball_x = 3
ball_y = 6
ball_dx = 0
ball_dy = 0
ball2bat = None
BALL_COL = (10, 20, 10)
ball_speed = 7

SCREEN_X = 8
SCREEN_Y = 8

joypad = gamezip.GAMEZIP()

loop = 0

while True:
    joypad.clear_screen()

    joypad.plot(ball_x, ball_y, BALL_COL)

    for dx in range(BAT_LEN):
        joypad.plot(bat_x + dx, BAT_Y, BAT_COL)

    joypad.show_screen()
    joypad.sleep(50)

    if ball_dx == 0 and ball_dy == 0:
        ball2bat = ball_x - bat_x
    else:
        ball2bat = None
[... 197 lines cut ...]
        ball_dy = -1
        ball_dx = random.randrange(2)
        if ball_dx == 0:
            ball_dx = -1
        joypad.reset_clock()
        

            
    loop += 1

2018-12-06 14:56:54,424 - mu.logic:645(restore_session) INFO: Loaded files.
2018-12-06 14:56:54,424 - mu.logic:649(restore_session) INFO: User defined environment variables: []
2018-12-06 14:56:54,424 - mu.logic:653(restore_session) INFO: Minify scripts on micro:bit? True
2018-12-06 14:56:54,452 - mu.logic:1110(change_mode) INFO: Workspace directory: /home/phil/mu_code
2018-12-06 14:57:00,239 - mu.modes.base:217(find_device) INFO: Found device on port: ttyACM0
2018-12-06 14:57:00,239 - mu.modes.base:218(find_device) INFO: Serial number: 9900023431634e45007940010000003800000000cca828bd
2018-12-06 14:57:00,354 - mu.modes.microbit:597(toggle_files) INFO: Toggle filesystem on.
2018-12-06 14:57:06,408 - mu.interface.panes:388(dropEvent) INFO: Getting 'k20k.txt' from micro:bit. Copying to '/home/phil/mu_code/k20k.txt'.
2018-12-06 14:57:07,803 - mu.modes.microbit:135(get) ERROR: not enough values to unpack (expected 2, got 1)
2018-12-06 14:57:07,899 - mu.interface.main:701(show_message) DEBUG: There was a problem getting 'k20k.txt' from the micro:bit. Please check Mu's logs for more information.
2018-12-06 14:57:07,900 - mu.interface.main:702(show_message) DEBUG: None
2018-12-06 14:58:24,782 - mu.logic:1037(show_admin) INFO: Showing logs from /home/phil/.cache/mu/log/mu.log
2018-12-06 14:58:55,480 - mu.logic:1037(show_admin) INFO: Showing logs from /home/phil/.cache/mu/log/mu.log
```
this is the problem `ERROR: not enough values to unpack (expected 2, got 1)` it's the same error as `ufs get k20k.txt` and `ufs get k20k.txt new_name.txt` produce

carlosperate, 2018-12-17:
Hi @rhubarbdog,

This might be a long shot, but do you still have the micro:bit with this file inside?
If so, could you use this tool to extract the flash contents of the micro:bit, save them as a hex file and attach them here? (There are windows and mac executables in the releases page) https://github.com/carlosperate/ubitflashtool

rhubarbdog, 2018-12-17:
No but i have recreated flashed 2 and the problem persists. I can't setup virtual environments so haven't tried to install `ubitflashtool` . i produced this hex file by typing `uflash main.py .` , `main.py` is an empty file i use for tab completion. The program has to be put on the microbit using 2 commands `uflash` and with microfs `ufs put test_fs.py main.py` . if i just use `uflash test_fs.py` i can't run the program to create the file it crashes with `OSError: 28`
/home/phil/projects/gamezip/micropython.hex.txt

rhubarbdog, 2018-12-17:
[micropython.hex.txt](https://github.com/mu-editor/mu/files/2685964/micropython.hex.txt)

zarify, 2019-06-18:
Hi - I noticed a similar problem recently, and it seems to do with the size of the file.
I was using a Micro:bit to log data to a CSV file and then transferring it back to my computer for analysis.
Around 8.5KB files I could transfer without a problem but any larger (same file name, same type of data, only size changed).
Could the Micro:bit be trying to load the entire file into memory before transmitting it back to Mu? That would fit in with my observations and sort of fits with the amount of memory that seems to be free at the time.