mu-editor/mu #718: Snap install mu-editor

TommyPujol06, 2018-12-09:
I installed mu by `snap install mu-editor`
When I try to run it via console `mu-editor` I get this 
```bash
Logging to /home/xxxxx/snap/mu-editor/common/.cache/mu/log/mu.log
This application failed to start because it could not find or load the Qt platform plugin "wayland-egl".

Available platform plugins are: eglfs, linuxfb, minimal, minimalegl, offscreen, xcb.

Reinstalling the application may fix this problem.
Aborted (core dumped)
 ```
I also tried to run it in GUI mode but it does not start :
[image]

I tried this :
```console
git clone https://github.com/mu-editor/mu.git
cd mu-master
pip3 install -r requirements.txt
```
And try it again but I get the same error.

[OS]: Ubuntu 18.04
[Python]: 3.6.7
[PIP]: 18.1

bennuttall, 2018-12-10:
@flexiondotorg any ideas?

TommyPujol06, 2018-12-19:
@bennuttall and @flexiondotorg I think I've solved this by installing xcb (It works so I solved it.): 
` sudo apt-get install xcb` 
You should add this to your snap "plugings". Also II woud like to remember the #720 pull request wich update's the license I will add the xcb too.