mu-editor/mu #725: Text selecting with UTF-8 characters - problem.

abixadamj, 2018-12-14:
Mu-Editor 1.0, Linux Mint 19 64-bit.
When selecting text, seems to be OK...

[image]

but when there is some character with UTF-8 (Polish special signs).. there is something wired..

[image]

It seems that if there is more special signs, then select is shifted...

[image]

tjguk, 2018-12-14:
Thanks for the description. For reproducibility would you mind posting a small example which fails in the way you describe, please? Either link to a gist or (if GH allows it) attach a code file here. Thanks

abixadamj, 2018-12-14:
```
from PyTechBrain import *
from time import sleep

'''
Polish letter in UTF-8: ą
'''

q = PyTechBrain()
```

tjguk, 2018-12-14:
Perfect; thanks. I'll dig into it

ntoll, 2018-12-17:
If you're quick... this'll make it into 1.0.2. ;-)

tjguk, 2018-12-17:
How quick do I have to be? (I don't think it's worth holding anything back for).

ntoll, 2018-12-17:
By the end of the week? My plan to cut a release today has been shot to pieces by other factors and so I'm just expanding the scope of what to include and hope to prepare for a Friday release on Thursday.

tjguk, 2018-12-17:
Ok; I'll see how that works out. But this is a minor malfunction of a little-used feature so...

ntoll, 2018-12-17:
Ack. ;-)

tjguk, 2018-12-18:
(laughs) https://github.com/mu-editor/mu/blob/master/mu/interface/editor.py#L348

ntoll, 2018-12-18:
Hah! :-)

tmontes, 2019-01-15:
Chiming in with my analysis:

* Culprit seems to be in `range_from_positions` called from `highlight_selected_matches` in https://github.com/mu-editor/mu/blob/master/mu/interface/editor.py where a `FIXME` note by @ntoll says:

  > Not clear whether the Scintilla conversions are expecting bytes or characters (ie codepoints).

* Looking at QScintilla's [QsciScintilla.lineIndexFromPosition](http://pyqt.sourceforge.net/Docs/QScintilla2/classQsciScintilla.html#a56aae6ef4721bae8016cce0b7beeb0a4) docs, one can read:

  > This will convert the **position byte index** to the *line line number and *index character index.

Failure motive:

* `highlight_selected_matches` is passing character offsets to `range_from_positions` which, will in turn, call `QsciScintilla.lineIndexFromPosition` which expects byte offsets.

* Everything works nicely as long as theres a one-to-one match between bytes and characters.

* Once that is no longer true, `QsciScintilla.lineIndexFromPosition` produces all sorts of funny result which were initially observed by @abixadamj.

I'll be issuing a preliminary PR sharing a possible solution. If deemed reasonable, I'll be happy improve it, adding tests, and making it mergeable.

PS: Is that the canonical URL for QScinttilla's documentation? Not 100% sure...

tmontes, 2019-03-29:
@abixadamj, Mu 1.1.0 alpha 1 should include the fix for the issue you raised. Can you please confirm that? You can grab it from [here](https://github.com/mu-editor/mu/releases/tag/1.1.0-alpha.1).

abixadamj, 2019-04-30:
@tmontes and @ntoll - I confirm - everything is OK ;-)

[image]

abixadamj, 2020-02-06:
@tmontes and @ntoll - Can you say, where 1.1.0 version will be available to use?

carlosperate, 2020-02-06:
The alpha is available in the downloads page: https://codewith.mu/en/download

carlosperate, 2020-09-07:
We can close this issue as the latest alpha ships with this fix.

closed 2020-09-07 by carlosperate