mu-editor/mu #731: Line Continuity "\" stops tune from playing in Adafruit CPX

V2man, 2019-01-03:
I am trying to play a longer tune with Adafruit Circuit Playground Express and Crickit board.
from adafruit_circuitplayground.express using 

 import cpx

notes = {'C': 262, 'D': 294, 'E': 330, 'F': 349, 'G': 391, 'A': 440, 'B': 494}
jingle = "EEE-EEE-EGCDE---FFFFFEEEEDDED-G-EEE-EEE-EGCDE---FFFFFEEEGGFDC---"
while True:
    for i in jingle:
         if i == '-':
            cpx.play_tone(2, 0.25)
        else:
            cpx.play_tone(notes[i], 0.25)

JamesLaderoute
 
Posts: 21
Joined: Wed Sep 14, 2016 10:38 pm on Adafruit Forum

I got  a long line exceeded 79 characters when I tried to extend the jingle variable beyond 79 characters. So I used "\" to indicate a continuity to the next line. I added a print(i) statement after the for i in jingle to see when the KeyError occurs. A KeyError occurs at the \ and stops the tune from finishing. 

jingle = "EEE-EEE-EGCDE---FFFFFEEEEDDED-G-\
              EEE-EEE-EGCDE---FFFFFEEEGGFDC---"
For example the above only plays the first line then the KeyError stops the program from running.

2019-01-03 00:20:53,328 - mu.interface.main:251(get_save_path) DEBUG: Getting save path: C:/Users/zrtur/Documents/Adafruit/mu_code/ConcurrentLoops.py
2019-01-03 00:20:53,330 - mu.logic:876(check_for_shadow_module) INFO: Checking path "C:/Users/zrtur/Documents/Adafruit/mu_code/ConcurrentLoops.py" for shadow module.
2019-01-03 00:20:53,330 - mu.logic:842(save_tab_to_file) INFO: Saving script to: C:/Users/zrtur/Documents/Adafruit/mu_code/ConcurrentLoops.py
2019-01-03 00:20:53,330 - mu.logic:843(save_tab_to_file) DEBUG: # Alternate tasks

MU 1.01 on WINDOWS 10 
 

Please remember to attach a **copy of the full log files for Mu**. You can get
the logs by clicking on the cog icon in the bottom right of the editor window.
Click on the logs and use CTRL-A to select all, then CTRL-C to copy and CTRL-V
to paste the contents into the issue.

Thank you for contributing to Mu! :-)

dlech, 2019-01-03:
> Sorry the \ I entered on the first line after ...DED-G-\ doesn't show up in the Preview above.

Tip: On GitHub, you can indent code lines 4 spaces or use a "code fence" like this:

````
```python
jingle = "EEE-EEE-EGCDE---FFFFFEEEEDDED-G- \
EEE-EEE-EGCDE---FFFFFEEEGGFDC---"
```
````

Then the code shows up like this:

```python
# This should work - note extra quotation marks
jingle = "EEE-EEE-EGCDE---FFFFFEEEEDDED-G-" \
    "EEE-EEE-EGCDE---FFFFFEEEGGFDC---"
```

(And if you missed the comment in the code above, read it again :wink: - it may solve your problem)

V2man, 2019-01-03:
Thank  you.  Putting quotes around each line or making use of the code fence solved the problem. I can now play longer tunes without getting the KeyError. 
I like number 2 below because I have more freedom to position the text from line to line for better readability. I am not restricted to location of the next tab.

1) Extra quotes must tab correctly on lines that follow first line. 
jingle = "EEE-EEE-EGCDE---FFFFFEEEEDDED-G-" \
                "EEE-EEE-EGCDE---FFFFFEEEGGFDC---"

2) Code fence more freedom to position text on each line
...
jingle = "EEE-EEE-EGCDE---FFFFFEEEEDDED-G- \
               EEE-EEE-EGCDE---FFFFFEEEGGFDC---"
...

3) Code fence with extra quotes also works.
...
jingle = "EEE-EEE-EGCDE---FFFFFEEEEDDED-G-" \
              "EEE-EEE-EGCDE---FFFFFEEEGGFDC---"
...

closed 2019-01-03 by V2man

dlech, 2019-01-03:
Interesting. The "code fence" was meant for writing on GitHub, not for use in Python.

V2man, 2019-01-04:
Are you saying a should not use the code fence ... in the MU editor?

dlech, 2019-01-04:
> Are you saying not use the code fence ... in the MU editor?

yes

> Do you know how to keep the preview window from removing my back slashes?

The code fence :smile:

dlech, 2019-01-04:
https://help.github.com/articles/basic-writing-and-formatting-syntax/