mu-editor/mu #732: CTRL-C does not break out of an infinite loop in the REPL

chumbers, 2019-01-03:
Using Mu 1.0.1 on Windows 10 in Python mode.
Start Mu
Click REPL
Paste this in the REPL
```
import random
names = ['Lisa', 'Kushal', 'Mary', 'Kate', 'Carlos', 'Zander', ]
while True:
    print(random.choice(names))
    wait = input()
```
Pres Enter a few times and see names printed randomly.

Press CTRL-C.  I expect to see the behavior described here (https://codewith.mu/en/tutorials/1.0/repl), which would be that it would break out of the infinite loop, but instead, nothing happens.

[mu.log](https://github.com/mu-editor/mu/files/2725575/mu.log)

TheKitty, 2019-01-09:
I had a similar situation yesterday, similar code but without input line. The Serial window appears to be acting up with Circuit Playground Express/Adafruit mode I believe.

lbeaussart, 2019-02-03:
Double click on REPL button will create a keyboard interruption.

dybber, 2020-02-15:
This is related to #808