mu-editor/mu #733: Feature request

MSWomack, 2019-01-04:
Is there a way to add the ability to predefine the data range for the plotter? Such as setting it to graph 0-50. right now it re adjusts to show the resent values, but goes from negative to positive. if I know the data I'm sending it will never be negative, for example, then half of the graph is wasted space. Another example is if I have a force sensor sending data, and I know that an impact is coming, and I'm showing the data graphed on a screen, it would be nice to have the range already set. Thank you.

dybber, 2020-06-13:
This is similar to  #969 and I've added a pull-request a few days #1059, which attempts to address this. Not by allowing users to define the range themselves, but by controlling the y-axis minimum independently of the y-axis maximum, and not adapting axis-changes as rapidly.

dybber, 2020-09-29:
The fix I mentioned (#1059) is now merged, which will correct the problem with wasted space in the graph (as explained above). I will thus close this issue.

closed 2020-09-29 by dybber