mu-editor/mu #734: No CircuitPython on Xubuntu 18.10

Meresmata, 2019-01-06:
If you are reporting a bug, we would like to know:

* I started mu 1.0.1 after installing it via pip3. After selecting Adafruit CircuitPython mu crashes. 
Meanwhile a Trinket-M0 were connected. 
     

The output in the console was:
#
Gtk-Message: Failed to load module "gail"
Gtk-Message: Failed to load module "atk-bridge"
Qt: Session management error: Authentication Rejected, reason : None of the authentication protocols specified are supported and host-based authentication failed
Traceback (most recent call last):
  File "/snap/mu-editor/4/bin/mu-editor", line 11, in <module>
    load_entry_point('mu-editor==1.0.1', 'console_scripts', 'mu-editor')()
  File "/snap/mu-editor/4/lib/python3.5/site-packages/mu/app.py", line 148, in run
    editor.restore_session(sys.argv[1:])
  File "/snap/mu-editor/4/lib/python3.5/site-packages/mu/logic.py", line 636, in restore_session
    self.select_mode(None)
  File "/snap/mu-editor/4/lib/python3.5/site-packages/mu/logic.py", line 1075, in select_mode
    self.change_mode(new_mode)
  File "/snap/mu-editor/4/lib/python3.5/site-packages/mu/logic.py", line 1112, in change_mode
    self.modes[mode].workspace_dir()))
  File "/snap/mu-editor/4/lib/python3.5/site-packages/mu/modes/adafruit.py", line 98, in workspace_dir
    mount_output = check_output(mount_command).splitlines()
  File "/snap/mu-editor/4/usr/lib/python3.5/subprocess.py", line 626, in check_output
    **kwargs).stdout
  File "/snap/mu-editor/4/usr/lib/python3.5/subprocess.py", line 693, in run
    with Popen(*popenargs, **kwargs) as process:
  File "/snap/mu-editor/4/usr/lib/python3.5/subprocess.py", line 947, in __init__
    restore_signals, start_new_session)
  File "/snap/mu-editor/4/usr/lib/python3.5/subprocess.py", line 1551, in _execute_child
    raise child_exception_type(errno_num, err_msg)
PermissionError: [Errno 13] Permission denied

#

OS: Xubuntu 18.10 with Kernel 4.18.0-13-generic on a Pentium N3450 (or so)

Thanks for helping out.

carlosperate, 2019-01-09:
Looks like a permission error when Mu launches a subprocess to try to detect any mounted drives (like the Trinket in this case):
https://github.com/mu-editor/mu/blob/717e2fe0e6db09896c3537003ecba03984f37da0/mu/modes/adafruit.py#L104-L108

It also looks like this is from a snap install, and I think this feature might not be supported yet: https://github.com/mu-editor/mu/issues/698#issuecomment-436277110

@bennuttall is this a matter of working on the snap config to enable access? Or are snaps disallowed to execute shell commands like this?

bennuttall, 2019-01-09:
Yes, likely. @flexiondotorg?

spacehobo, 2020-05-05:
I just got hit by this, and it seems to be a permission that could be added: 
```
May  5 12:16:03 quelbo kernel: [654358.200770] audit: type=1400 audit(1588677363.549:7666534): apparmor="DENIED" operation="open" profile="snap.mu-editor.mu-editor" name="/proc/1656553/mounts" pid=1656553 comm="python3" requested_mask="r" denied_mask="r" fsuid=1000 ouid=1000
May  5 12:16:06 quelbo kernel: [654361.147045] audit: type=1400 audit(1588677366.493:7666539): apparmor="DENIED" operation="open" profile="snap.mu-editor.mu-editor" name="/run/udev/data/c4:28" pid=1656553 comm="python3" requested_mask="r" denied_mask="r" fsuid=1000 ouid=0
```
It's difficult to see what went wrong, because the `snapcraft.yaml` file isn't included in the main repo where it belongs.  Is there any chance that could be added so we could submit PRs for it?

bennuttall, 2020-05-05:
The `snapcraft.yml` is in an old branch https://github.com/mu-editor/mu/tree/snap

I started the snap effort, but didn't really keep up with it. I think it could be worth pursuing but I'd need to learn more about how snaps work, and I don't really have the time.