mu-editor/mu #735: dev-setup: `make pycodestyle` should not dive into `venv`

wayoda, 2019-01-08:
This relates to the developer setup on **Linux** *(Ubuntu 18.10)*
 
I cloned the mu-repo and installed the virtual environment in `venv`

Running `make pycodestyle` takes **4 minutes** to complete.
The reason it takes so long is that the `find` command prepended to the actual test unnecessarily dives into the `venv` directory to collect all python source files. The `pycodestyle` command itself discards any source files originating from `venv`. 
When the `find` command is forced to ignore `venv` ,  `make pycodestyle` completes in **6 seconds**    

Here is an excerpt from `git diff` that shows the changes I made to the `Makefile`
```
 pycodestyle:
-       find . \( -name _build -o -name var \) -type d -prune -o -name '*.py' -print0 | $(XARGS) -n 1 pycodestyle --repeat --exclude=build/*,docs/*,mu/contrib*,mu/modes/api/*,utils/*,venv/*,.vscode/* --ignore=E731,E402,W504
+       find . \( -name _build -o -name var -o -path ./venv \) -type d -prune -o -name '*.py' -print0 | $(XARGS) -n 1 pycodestyle --repeat --exclude=build/*,docs/*,mu/contrib*,mu/modes/api/*,utils/*,venv/*,.vscode/* --ignore=E731,E402,W504
```

dlech, 2019-01-08:
It would be nice to add `env/` here too since that is another common place to store a virtualenv.

wayoda, 2019-01-08:
You run into more trouble if you use `env` for the virtual environment. `make pyflakes` already ignores `venv` in the current configuration. But it dives into `env` and tests the pep8 conformity of all the  python packages intalled by the virtual env. Interesting to see how many packages are not conforming to pep8, but you really don't want to see hundreds of warning messages scroll by... 

A better idea would probably be to update the developer docs. Currently it says: 
*From this point onwards, instructions assume that you’re using a virtual environment.*
This should be updated to:
*From this point onwards, instructions assume that you’re using a virtual environment **created in `venv`**.*

dlech, 2019-01-08:
FWIW, I've been using `env/` since I use VS Code for development. It will automatically find a virtualenv in `env/` but not `venv/`. It is not too hard to manually configure it to use `venv/` instead though. It is just slightly easier to the automatic way.

wayoda, 2019-01-08:
Everybody got his prefered way to do things. I use [Pipenv](https://pipenv.readthedocs.io/en/latest/) which puts the virtual env into a special directory under the users homedir *not* into a subdir of `mu`.
@dlech did you experience the same problems I had with using `env` in combination with `make pycodestyle`?

dlech, 2019-01-08:
yes, this is the change I made:

```diff
@@ -48,10 +48,10 @@ else
 endif
 
 pyflakes:
-	find . \( -name _build -o -name var -o -path ./docs -o -path ./mu/contrib -o -path ./utils -o -path ./venv \) -type d -prune -o -name '*.py' -print0 | $(XARGS) pyflakes
+	find . \( -name _build -o -name var -o -path ./docs -o -path ./mu/contrib -o -path ./utils -o -path ./venv -o -path ./env \) -type d -prune -o -name '*.py' -print0 | $(XARGS) pyflakes
 
 pycodestyle:
-	find . \( -name _build -o -name var \) -type d -prune -o -name '*.py' -print0 | $(XARGS) -n 1 pycodestyle --repeat --exclude=build/*,docs/*,mu/contrib*,mu/modes/api/*,utils/*,venv/*,.vscode/* --ignore=E731,E402,W504
+	find . \( -name _build -o -name var -o -name env \) -type d -prune -o -name '*.py' -print0 | $(XARGS) -n 1 pycodestyle --repeat --exclude=build/*,docs/*,mu/contrib*,mu/modes/api/*,utils/*,venv/*,.vscode/* --ignore=E731,E402,W504
 
 test: clean
 	pytest
```

It seems to be working well, but I just noticed that it looks like I missed adding `,env/*` in the `--exclude=` option.

wayoda, 2019-01-08:
Adding `env` to the `--exclude` arg is not neccesary because `find` collects the `*.py` files and pipes them to `pycodestyle`.  If `find` ignores `env` `pycodestyle` will never see a file from that directory.

wayoda, 2019-10-10:
Closing this issue since the make process has been updated and the `find` command isn't used any more

closed 2019-10-10 by wayoda