mu-editor/mu #738: Program stops before completing loop

martinohanlon, 2019-01-08:
Fyi - I suspect this is an issue which is already raised, or at least a solution to the problem will resolve more than 1 issue, it certainly seems similar to #510 .

When executing programs which contain fast loops, sometimes the program doesnt finish, I have seen this happen in the "real world", but you can replicate it with a simple `for` loop.

e.g. Running

```python
for i in range(1000):
    print(i)
```

will usually result in a value between `150` & `175` being printed before stopping.

[image]

What I find curious though, is that if I run the same program with a higher range e.g.

```python
for i in range(10000):
    print(i)
```

The program will for longer generally to between `1225` and `1250`.

[image]

There is nothing reported in the mu log, just that the script started and then when you eventually click Stop, that the script was stopped.

```
2019-01-08 14:03:01,890 - mu.modes.python3:191(run_script) DEBUG: for i in range(10000):

    print(i)

2019-01-08 14:03:01,893 - mu.interface.panes:644(start_process) INFO: Running script: c:\users\martin o'hanlon-lt\mu_code\test_buffer.py
2019-01-08 14:03:01,893 - mu.interface.panes:646(start_process) INFO: Running with interactive mode.
2019-01-08 14:03:01,893 - mu.interface.panes:649(start_process) INFO: Command args: []
2019-01-08 14:03:01,893 - mu.interface.panes:667(start_process) INFO: Working directory: C:\Users\Martin O'Hanlon-LT\mu_code
2019-01-08 14:03:01,893 - mu.interface.panes:672(start_process) INFO: Python path: ["C:\\Users\\Martin O'Hanlon-LT\\AppData\\Local\\Mu\\pkgs", "C:\\Users\\Martin O'Hanlon-LT\\AppData\\Local\\Mu\\Python\\python36.zip", "C:\\Users\\Martin O'Hanlon-LT\\AppData\\Local\\Mu\\Python", "C:\\Users\\Martin O'Hanlon-LT\\AppData\\Local\\Mu\\pkgs", "C:\\Users\\Martin O'Hanlon-LT\\AppData\\Local\\Mu\\pkgs\\IPython\\extensions"]
2019-01-08 14:03:03,322 - mu.logic:1039(show_admin) INFO: Showing logs from C:\Users\Martin O'Hanlon-LT\AppData\Local\python\mu\Logs\mu.log
```

This was using mu 1.0.1 and the behaviour is consistent on both Windows 10 and Raspbian (on a Raspberry Pi).

martinohanlon, 2019-01-08:
Introducing the smallest of delays results in the programming running succesfully.

```python
from time import sleep
for i in range(10000):
    print(i)
    sleep(0.0000001)
```

[image]

carlosperate, 2019-01-08:
Yes, I think this should have been resolved with https://github.com/mu-editor/mu/pull/723.
If you'd like to test it you can try running from master, or one of the latest builds from http://mu-builds.s3-website.eu-west-2.amazonaws.com/?prefix=windows/.

martinohanlon, 2019-01-08:
Thanks a lot @carlosperate .

Tested and confirmed no longer an issue.

I will close this issue as it is already well documented.

closed 2019-01-08 by martinohanlon