mu-editor/mu #740: Mu should handle unknown files when checking for duplicates.

ntoll, 2019-01-15:
```
2019-01-15 11:36:36,334 - root:95(excepthook) ERROR: Unrecoverable error
Traceback (most recent call last):
  File "C:\Program Files (x86)\Mu\pkgs\mu\logic.py", line 831, in load
    self._load(path)
  File "C:\Program Files (x86)\Mu\pkgs\mu\logic.py", line 730, in _load
    if os.path.samefile(path, widget.path):
  File "genericpath.py", line 97, in samefile
FileNotFoundError: [WinError 3] The system cannot find the path specified: 'D:/main.py'
```

Mu had a file open from a subsequently unplugged CircuitPlaygroundExpress (i.e. the file was no longer accessible).

tjguk, 2019-01-15:
Do you need any Win-specific help here, @ntoll? os.path.samefile on windows opens both files and compares their handles; roughly equivalent to comparing inodes on *nix. But looks like an exception handler here is probably what you'll be going for...

ntoll, 2019-01-15:
@tjguk that's exactly the conclusion I came to. The chances of this bug ever rearing its head are slim, but I wanted to log it so it's fixed in the *next* version or Mu. ;-)

All Windows builds have passed muster. Just looking at OSX for the 1.0.2 release.

carlosperate, 2021-08-04:
@ntoll is this still an issue in Windows or Linux? I haven't been able to replicate in macOS and get this error dialog instead:

[image]