mu-editor/mu #743: mu 1.0.2 Can't open directly from launchpad on macOS

JohnGuan, 2019-01-18:
Can't open directly from Launchpad on macOS, but can start in Terminal.
macOS version: 10.14.2 (18C54)
`2019-01-18 11:18:28,589 - root:95(excepthook) ERROR: Unrecoverable error
Traceback (most recent call last):
  File "/Applications/mu-editor.app/Contents/MacOS/mu-editor", line 20, in <module>
    load_entry_point('mu-editor==1.0.2', 'console_scripts', 'mu-editor')()
  File "/Applications/mu-editor.app/Contents/Resources/app/mu/app.py", line 111, in run
    setup_logging()
  File "/Applications/mu-editor.app/Contents/Resources/app/mu/app.py", line 66, in setup_logging
    print(_('Logging to {}').format(LOG_FILE))
UnicodeEncodeError: 'ascii' codec can't encode characters in position 0-4: ordinal not in range(128)
2019-01-18 11:18:49,586 - root:95(excepthook) ERROR: Unrecoverable error
Traceback (most recent call last):
  File "/Applications/mu-editor.app/Contents/MacOS/mu-editor", line 20, in <module>
    load_entry_point('mu-editor==1.0.2', 'console_scripts', 'mu-editor')()
  File "/Applications/mu-editor.app/Contents/Resources/app/mu/app.py", line 111, in run
    setup_logging()
  File "/Applications/mu-editor.app/Contents/Resources/app/mu/app.py", line 66, in setup_logging
    print(_('Logging to {}').format(LOG_FILE))
UnicodeEncodeError: 'ascii' codec can't encode characters in position 0-4: ordinal not in range(128)
2019-01-18 11:20:42,166 - root:95(excepthook) ERROR: Unrecoverable error
Traceback (most recent call last):
  File "/Applications/mu-editor.app/Contents/MacOS/mu-editor", line 20, in <module>
    load_entry_point('mu-editor==1.0.2', 'console_scripts', 'mu-editor')()
  File "/Applications/mu-editor.app/Contents/Resources/app/mu/app.py", line 111, in run
    setup_logging()
  File "/Applications/mu-editor.app/Contents/Resources/app/mu/app.py", line 66, in setup_logging
    print(_('Logging to {}').format(LOG_FILE))
UnicodeEncodeError: 'ascii' codec can't encode characters in position 0-4: ordinal not in range(128)
2019-01-18 11:21:20,405 - root:95(excepthook) ERROR: Unrecoverable error
Traceback (most recent call last):
  File "/Applications/mu-editor.app/Contents/MacOS/mu-editor", line 20, in <module>
    load_entry_point('mu-editor==1.0.2', 'console_scripts', 'mu-editor')()
  File "/Applications/mu-editor.app/Contents/Resources/app/mu/app.py", line 111, in run
    setup_logging()
  File "/Applications/mu-editor.app/Contents/Resources/app/mu/app.py", line 66, in setup_logging
    print(_('Logging to {}').format(LOG_FILE))
UnicodeEncodeError: 'ascii' codec can't encode characters in position 0-4: ordinal not in range(128)
2019-01-18 11:21:51,573 - root:95(excepthook) ERROR: Unrecoverable error
Traceback (most recent call last):
  File "/Applications/mu-editor.app/Contents/MacOS/mu-editor", line 20, in <module>
    load_entry_point('mu-editor==1.0.2', 'console_scripts', 'mu-editor')()
  File "/Applications/mu-editor.app/Contents/Resources/app/mu/app.py", line 111, in run
    setup_logging()
  File "/Applications/mu-editor.app/Contents/Resources/app/mu/app.py", line 66, in setup_logging
    print(_('Logging to {}').format(LOG_FILE))
UnicodeEncodeError: 'ascii' codec can't encode characters in position 0-4: ordinal not in range(128)
2019-01-18 11:22:03,638 - root:95(excepthook) ERROR: Unrecoverable error
Traceback (most recent call last):
  File "/Applications/mu-editor.app/Contents/MacOS/mu-editor", line 20, in <module>
    load_entry_point('mu-editor==1.0.2', 'console_scripts', 'mu-editor')()
  File "/Applications/mu-editor.app/Contents/Resources/app/mu/app.py", line 111, in run
    setup_logging()
  File "/Applications/mu-editor.app/Contents/Resources/app/mu/app.py", line 66, in setup_logging
    print(_('Logging to {}').format(LOG_FILE))
UnicodeEncodeError: 'ascii' codec can't encode characters in position 0-4: ordinal not in range(128)
2019-01-18 11:22:21,222 - root:95(excepthook) ERROR: Unrecoverable error
Traceback (most recent call last):
  File "/Applications/mu-editor.app/Contents/MacOS/mu-editor", line 20, in <module>
    load_entry_point('mu-editor==1.0.2', 'console_scripts', 'mu-editor')()
  File "/Applications/mu-editor.app/Contents/Resources/app/mu/app.py", line 111, in run
    setup_logging()
  File "/Applications/mu-editor.app/Contents/Resources/app/mu/app.py", line 66, in setup_logging
    print(_('Logging to {}').format(LOG_FILE))
UnicodeEncodeError: 'ascii' codec can't encode characters in position 0-4: ordinal not in range(128)
2019-01-18 11:25:53,070 - root:95(excepthook) ERROR: Unrecoverable error
Traceback (most recent call last):
  File "/Applications/mu-editor.app/Contents/MacOS/mu-editor", line 20, in <module>
    load_entry_point('mu-editor==1.0.2', 'console_scripts', 'mu-editor')()
  File "/Applications/mu-editor.app/Contents/Resources/app/mu/app.py", line 111, in run
[... 191 lines cut ...]

-----------------

Starting Mu 1.0.2
2019-01-18 14:15:50,222 - root:113(run) INFO: uname_result(system='Darwin', node='guanninuandeMBP.lan', release='18.2.0', version='Darwin Kernel Version 18.2.0: Mon Nov 12 20:24:46 PST 2018; root:xnu-4903.231.4~2/RELEASE_X86_64', machine='x86_64', processor='i386')
2019-01-18 14:15:50,222 - root:114(run) INFO: Python path: ['/Applications/mu-editor.app/Contents/MacOS', '/Applications/mu-editor.app/Contents/Resources/python/lib/python36.zip', '/Applications/mu-editor.app/Contents/Resources/python/lib/python3.6', '/Applications/mu-editor.app/Contents/Resources/python/lib/python3.6/lib-dynload', '/Applications/mu-editor.app/Contents/Resources/python/lib/python3.6/site-packages', '/Applications/mu-editor.app/Contents/Resources/app', '/Applications/mu-editor.app/Contents/Resources/app_packages', '/Applications/mu-editor.app/Contents/Resources/app_packages/IPython/extensions']
2019-01-18 14:15:50,222 - root:115(run) INFO: Language code: zh_CN
2019-01-18 14:15:50,515 - mu.logic:543(__init__) INFO: Setting up editor.
2019-01-18 14:15:50,516 - mu.logic:562(__init__) INFO: Settings path: /Users/johnguan/Library/Application Support/mu/settings.json
2019-01-18 14:15:50,516 - mu.logic:563(__init__) INFO: Session path: /Users/johnguan/Library/Application Support/mu/session.json
2019-01-18 14:15:50,516 - mu.logic:564(__init__) INFO: Log directory: /Users/johnguan/Library/Logs/mu
2019-01-18 14:15:50,516 - mu.logic:565(__init__) INFO: Data directory: /Users/johnguan/Library/Application Support/mu
2019-01-18 14:15:50,529 - mu.logic:578(setup) INFO: Available modes: python, adafruit, microbit, debugger, pygamezero
2019-01-18 14:15:50,680 - mu.logic:1150(change_mode) INFO: Workspace directory: /Users/johnguan/mu_code
2019-01-18 14:15:50,707 - mu.logic:627(restore_session) INFO: Restoring session from: /Users/johnguan/Library/Application Support/mu/session.json
2019-01-18 14:15:50,707 - mu.logic:628(restore_session) DEBUG: {'theme': 'day', 'mode': 'microbit', 'paths': [], 'envars': [], 'minify': False, 'microbit_runtime': '', 'zoom_level': 2}
2019-01-18 14:15:50,707 - mu.logic:649(restore_session) INFO: Loaded files.
2019-01-18 14:15:50,707 - mu.logic:653(restore_session) INFO: User defined environment variables: []
2019-01-18 14:15:50,707 - mu.logic:657(restore_session) INFO: Minify scripts on micro:bit? False
2019-01-18 14:15:50,803 - mu.logic:680(restore_session) INFO: Starting with blank file.
2019-01-18 14:15:50,815 - mu.logic:1150(change_mode) INFO: Workspace directory: /Users/johnguan/mu_code
2019-01-18 14:15:54,536 - mu.logic:1077(show_admin) INFO: Showing logs from /Users/johnguan/Library/Logs/mu/mu.log
2019-01-18 16:12:11,529 - mu.logic:1077(show_admin) INFO: Showing logs from /Users/johnguan/Library/Logs/mu/mu.log
2019-01-18 16:17:49,659 - mu.logic:1077(show_admin) INFO: Showing logs from /Users/johnguan/Library/Logs/mu/mu.log
`

ntoll, 2019-01-18:
@tmontes this didn't used to happen. Could it be something to do with the changes you made to OSX locale detection? (This is just a hunch based on what has/hasn't changed since the last release.)

ntoll, 2019-01-18:
OK... here's what I've learned.

Previously, because of the odd way OSX does locale, it was falling back to "English" as the default language, so the `print` statement on line 66 was getting the string `"Logging to ..."`. Now that Mu correctly detects OSX locale settings (thanks @tmontes), the `print` statement is getting the text translated into Chinese (presumably your locale): `"记录日志到 /Users/johnguan/Library/Logs/mu/mu.log"` 

At this point, Python complains about encoding problems and breaks.

tmontes, 2019-01-18:
@ntoll, thanks for pinging me. :)

I'll definitely try to have a look at this sometime later today. A few notes off the top of my head:
* Your reasoning makes sense: the `print` function is failing at printing.
* The string should indeed be the `zh_CN` locale translated `'Logging to ...'`: `'记录日志到 ...'`.
* The `print` failure results from failing to encode such a string to the `sys.stdout` encoding.
* I know that recent CPython versions have changed/improved the way `sys.stdout` encoding is determined and that, by itself, it is a non-trivial problem.
* It most probably is not UTF-8 as the encoding process should not fail with it... Maybe some fallback code in CPython "decided" to set `sys.stdout.encoding` to ASCII?

Pseudo-reproduction example:
```shell
$ python3.6
Python 3.6.8 (v3.6.8:3c6b436a57, Dec 24 2018, 02:04:31) 
[GCC 4.2.1 Compatible Apple LLVM 6.0 (clang-600.0.57)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> import sys
>>> sys.stdout.encoding
'UTF-8'
>>> s = '记录日志到'
>>> print(s)             # works, `s` successfully encoded to UTF-8
记录日志到
>>> s.encode('UTF-8')    # confirmation
b'\xe8\xae\xb0\xe5\xbd\x95\xe6\x97\xa5\xe5\xbf\x97\xe5\x88\xb0'
>>> s.encode('ASCII')    # pseudo-reproduction
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
UnicodeEncodeError: 'ascii' codec can't encode characters in position 0-4: ordinal not in range(128)
>>>
```

I get the same exact behaviour on CPython 3.7 here.

Suspicion: the CPython startup process is setting `sys.stdout.encoding` to something that can't encode the localised message. Needs further investigation which I'll try to do later today.

tmontes, 2019-01-18:
Progress - Confirmed suspicion above.

What I did:
* Using mu-editor.app 1.0.2 from the distributed DMG.
* Installed in /Applications/Local.
* Can be launched from macOS's launchpad (my system: macOS Sierra 10.12.6).
* Hand edited "/Applications/Local/mu-editor.app/Contents/Resources/app/mu/app.py":
  * Under `setup_logging` just before the `print(_('Logging to {}').format(LOG_FILE))` line:
  * Added `log.info('sys.stdout.encoding=%r', sys.stdout.encoding)`
* Removed all mu log files in ~/Library/Logs/mu.
* Launched mu via macOS's launchpad. Log file contents:
  `2019-01-18 10:47:32,263 - root:66(setup_logging) INFO: sys.stdout.encoding='US-ASCII'`
* Launched from the terminal with `open /Applications/Local/mu-editor.app`. Log file contents:
  `2019-01-18 10:52:54,605 - root:66(setup_logging) INFO: sys.stdout.encoding='UTF-8'`

Possible quick fixes:
* Just don't `print` things.
* If `print` is useful, probably for diagnostic purposes`, don't localize `print`ed things.

"Real" fixes:
* Possibly not easy given that CPython devs have been working in recent versions.
* Clearly needs further investigation.
* Hard-fact: no-one can ever guarantee that a given process' STDOUT (which is a byte stream) can handle UTF-8 bytes... So maybe one of the quick-fix approaches I mentioned (or some variation?) is the only safe way forward.

Plan on doing further investigation:
* Recent Python changes in detecting/setting STDIN/OUT encoding.
* macOS's process environment for processes launched from launchpad.

tmontes, 2019-01-18:
@JohnGuan,

Do you think you could "hack" your mu installation -- with our guidance -- to see if we can make it work for you, when launched from the launchpad? This would be helpful in confirming the root cause of the issue on your system.

The idea would be:
* Edit the `Contents/Resources/app/mu/app.py` inside the mu-editor.app bundle.
* Change line 66 from...
  `print(_('Logging to {}').format(LOG_FILE))`
  to
  `print('Logging to {}'.format(LOG_FILE))`
* Add a diagnostic line, just before that one with:
  `log.info('sys.stdout.encoding=%r', sys.stdout.encoding)`

If this is something you feel you could try then great:
* Please change the file and then try launching mu from the launchpad.
* It will hopefully be successful.
* The log file in `~/Library/Logs/mu/...` should include a line with saying `sys.stdout.encoding=...`... Please share that.

Otherwise, if this feels to complex don't worry. I'm sure we'll be able to make it work otherwise.

ntoll, 2019-01-18:
@tmontes thank you! What you've just written on this issue is exactly my reading of the situation. I have already walked @JohnGuan through similar steps (although I omitted to ask for sys.stdout.encoding).

I believe the simplest solution is simply to remove the `print` statement. We only use `print` once and it's only there because I didn't remove it when I added the "log" pane to the UI (it helped sys admins at school find out what Mu was complaining about).

Ergo, I'll remove `print` and update the installers with the change.

tmontes, 2019-01-18:
(now away from laoptop/code but...)

While you are at it, it may make sense to remove (or wrap in a try/except UnicodeEncodingError) any other `print` calls the code may have lying around... I have the feeling I saw at least one more. :)

ntoll, 2019-01-18:
@tmontes I had a look at the use of `print` in the code base. We don't have any further uses which use gettext (i.e. all other print statements are in "contrib" modules which are outputting English ASCII informational messages.

tmontes, 2019-01-18:
@ntoll that's good... :) But won't [this one here, `app.py:debug()`](https://github.com/mu-editor/mu/blob/master/mu/app.py#L183) be potentially problematic?

ntoll, 2019-01-18:
How did I miss that? Doh. You have very sharp eyes.

Yes... I agree. That could be problematic. I'll re-open.

tmontes, 2019-01-18:
...I'm not sure under which circumstances the `print` in `app.py:debug()` would run -- IOW: how would it be missing the module to be debugged? Regardless of that, even if the argument isn't present, how would that `print` output reach a potential user? Is it captured somehow and displayed in a graphical pane? Or is the debugger supposed to be runnable from a command line interface as well, for more advanced users?

ntoll, 2019-01-18:
The way the debugger works is described here: https://mu.readthedocs.io/en/latest/debugger.html#graphical-debugger

Basically, this entry point is used by the DebugRunner side of things to launch the code to be debugged. The output from this process is displayed in Mu's run pane - used to show and capture stdin/stdout.

Does this make sense?

I agree it may need a try/except around it though, for the same reason we removed the `print` statement.

Having said that, the ONLY way this block of code could be reached is if the user launched the DebugRunner "manually" from the command line WITHOUT specifying the python file to be debugged. Mu *always* launches the DebugRunner with a filename (it checks/complains if there's no filename), so "normal" users will never encounter this code branch. The only person on the planet who is ever likely to launch the DebugRunner "manually" is me, for development purposes, and I don't speak Chinese. ;-)

If you're happy that the chances of a Chinese speaking developer manually running the DebugRunner are sufficiently small (i.e. I think the chances are basically zero), then I'm happy to close this issue again. However, I can't help but smile to myself and think this sort of attitude is bound to come back and haunt me in the future. ;-)

tjguk, 2019-01-18:
FWIW I agree with @tmontes in principle.

But also I'd add what is doubtless obvious to both of you: that having the discussion and thought process "out loud" as it were is so helpful later when you can't quite remember what process you went through to get to that point. (Even if you dutifully put good comments in place).

That's where I'm haltingly trying to go with my design notes idea:

https://github.com/mu-editor/mu/tree/master/docs/design

ntoll, 2019-01-18:
+1 to the assessment, discussion and suggested solution. Also +1 to public discussion about such things. Thank you @tmontes and @tjguk. I'll fix right now.

closed 2019-01-18 by ntoll

tmontes, 2019-01-18:
Thanks @ntoll + @tjguk for your inputs and guidance. :)

commit 3eaa305b (2019-01-18): Fixes #743, a guard against encoding problems with the print statement.