mu-editor/mu #746: macOS | pygame zero | Fails running with non-ASCII comment

tmontes, 2019-01-23:
## Preabmle

* This is not *strictly* a Mu bug.
* I narrowed it down to a Pygame Zero behaviour that could/should be improved:
  * I'll log and link an issue in the respective issue tracker.
* Nevertheless, once fixed in Pygame Zero, Mu distributions will need to bundle an updated version.
* I also decided to log the issue here to share my analysis and workaround with other Mu users.

## Platform Notes

* Does fail on macOS.
* Does not fail on Windows.
* Not sure if it fails on Linux, but I guess it might not.

## Steps to Reproduce on macOS

* Launch Mu, installed from the [distributed DMG](
https://github.com/mu-editor/mu/releases/download/1.0.2/mu-editor_1.0.2_osx.dmg), via the Finder or LaunchPad.
  (launching from the terminal does not lead to the failure, explanation below)
* Set Mu mode to **Pygame Zero**.
* Create new file with:

```python
WIDTH = 300
HEIGHT = 200
```

* Save it as "pgz_failure.py".
* Click **Play**.
* Pygame Zero runs the file successfully.
* Click **Stop**.
* Edit the "pgz_failure.py" file by adding a comment containing non-ASCII characters like:

```python
# Résumé
WIDTH = 300
HEIGHT = 200
```

* Save and click **Play**.
* Running fails with the following output on the lower *Running* pane:

```
pygame 1.9.4
Hello from the pygame community. https://www.pygame.org/contribute.html
Traceback (most recent call last):
  File "/Users/travis/build/mu-editor/mu_portable_python_macos/python/lib/python3.6/runpy.py", line 193, in _run_module_as_main
  File "/Users/travis/build/mu-editor/mu_portable_python_macos/python/lib/python3.6/runpy.py", line 85, in _run_code
  File "/Applications/Local/mu-editor.app/Contents/Resources/app_packages/pgzero/__main__.py", line 3, in <module>
    main()
  File "/Applications/Local/mu-editor.app/Contents/Resources/app_packages/pgzero/runner.py", line 77, in main
    src = f.read()
  File "/Users/travis/build/mu-editor/mu_portable_python_macos/python/lib/python3.6/encodings/ascii.py", line 26, in decode
UnicodeDecodeError: 'ascii' codec can't decode byte 0xc3 in position 3: ordinal not in range(128)

---------- FINISHED ----------
exit code: 1 status: 0
```

* Click **Stop**.
* No relevant details are captured in the log file other than:
```
2019-01-23 14:17:34,485 - mu.modes.pygamezero:129(run_game) DEBUG: # Résumé
WIDTH = 300
HEIGHT = 200
2019-01-23 14:17:34,493 - mu.interface.panes:651(start_process) INFO: Running script: /Users/tiago.montes/mu_code/pgz_failure.py
2019-01-23 14:17:34,493 - mu.interface.panes:656(start_process) INFO: Command args: []
2019-01-23 14:17:34,493 - mu.interface.panes:714(start_process) INFO: Working directory: /Users/tiago.montes/mu_code
2019-01-23 14:17:34,493 - mu.interface.panes:719(start_process) INFO: Python path: ['/Applications/Local/mu-editor.app/Contents/MacOS', '/Applications/Local/mu-editor.app/Contents/Resources/python/lib/python36.zip', '/Applications/Local/mu-editor.app/Contents/Resources/python/lib/python3.6', '/Applications/Local/mu-editor.app/Contents/Resources/python/lib/python3.6/lib-dynload', '/Applications/Local/mu-editor.app/Contents/Resources/python/lib/python3.6/site-packages', '/Applications/Local/mu-editor.app/Contents/Resources/app', '/Applications/Local/mu-editor.app/Contents/Resources/app_packages', '/Applications/Local/mu-editor.app/Contents/Resources/app_packages/IPython/extensions']
2019-01-23 14:17:38,831 - mu.modes.pygamezero:143(stop_game) DEBUG: Stopping script.
```

## Analisys

Failure stems from the following code snippet in **Pygame Zero**'s `runner.py` code, in the `main` function, at lines 76-77 in the distributed file, but [slightly moved in the current master here](https://github.com/lordmauve/pgzero/blob/master/pgzero/runner.py#L102):

```python
    with open(path) as f:
        src = f.read()                                                                    
```

* The culprit is the exception raised by the `f.read()` call, stating that the `ascii` codec can't decode the file bytes.
* The `open` function omits the `mode` and `encoding` arguments, meaning, in Python 3:
  * Use `rt` mode: read text.
  * Use whatever `locale.getpreferredencoding(False)` returns as the encoding to decode file bytes to strings.
* Much like in #743 (where a failure was reported when running Mu on macOS when launched from the LaunchPad) both the Mu and the Pygame Zero Python processes get `US-ASCII` from the `locale.getpreferredencoding(False)` call.
* Underlying motive: neither the Finder, nor LaunchPad set one of the many environment variables that would make that call return something like `UTF-8`. That's why launching Mu from the terminal does not lead to the failure (the terminal sets the LC_CTYPE environment variable to UTF-8 which makes the problem go away).

## Workaround

* Bring up the "Mu Administration" window by clicking the cog wheel on the right side of the bottom status bar.
* Under the "Python3 Environment" tab add a line like:

```
LC_CTYPE=UTF-8
```

* Click the "Ok" button.
* The Pygame Zero program with non-ASCII comments should now run successfully.

## The real fix

AFAICT, the fix should first be applied to Pygame Zero. The idea is simple:

* The code I quoted above, reads the `*.py` module file into the `src` variable as an `str` object.
* It then passes `str` to the built-in `compile` function, preparing its execution.

Given that `compile` accepts not only `str` objects, but also `bytes` and AST objects -- [reference docs here](https://docs.python.org/3/library/functions.html#compile) -- the fix could be as simple as changing the `open` call to use `mode='rb'` instead of omitting it; in such case, the `str` object returned by `f.read()` would be of type `bytes` with no decoding taking place and, thus, no room for `UnicodeDecodeError`s. :-)

I tried it out in my installation and it worked nicely with a couple of test projects.

## Next

* I'll log the issue in Pygame Zero's issue tracker and link it back here.
* Would love to hear other people's perspective on this.

ntoll, 2019-01-23:
Hi @tmontes, thanks for reporting this. Something for @lordmauve to take a look into perhaps.

tjguk, 2019-01-23:
Well, my perspective is: thanks for picking this up and doing the detective work to narrow it down. Someone is bound to be bitten by this in a classroom situation so it's great to (a) know about it and hopefully have an upstream fix in the pipeline; and (b) have a workaround on tap just in case.

FWIW I agree with your analysis (having spent a certain amount of time doing the equivalent work around Mu itself).

tmontes, 2019-05-08:
Quick update: @lordmauve has just updated pgzero with a fix to this issue. As soon as there's an updated release on PyPI we can update Mu's pgzero version dependency and fix this. :)

carlosperate, 2020-11-08:
Okay, so it sounds like this will be fixed by updating pygamezero. When the dependencies are updated for the next alpha/beta release we can check this.

ntoll, 2021-07-10:
Since we're using the updated PyGameZero, closing this.

closed 2021-07-10 by ntoll