mu-editor/mu #747: make check fails on 'pyflakes' target

ShawnHymel, 2019-01-25:
```
$ make check
rm -rf build
rm -rf dist
rm -rf mu_editor.egg-info
rm -rf .coverage
rm -rf .eggs
rm -rf docs/_build
rm -rf .pytest_cache
rm -rf lib
rm -rf pynsist_pkgs
rm -rf pynsist_tkinter*
rm -rf macOS
rm -rf *.mp4
rm -rf .git/avatar/*
find . \( -name '*.py[co]' -o -name dropin.cache \) -delete
find . \( -name '*.bak' -o -name dropin.cache \) -delete
find . \( -name '*.tgz' -o -name dropin.cache \) -delete
find . | grep -E "(__pycache__)" | xargs rm -rf
find . \( -name _build -o -name var \) -type d -prune -o -name '*.py' -print0 | xargs -0 -r -n 1 pycodestyle --repeat --exclude=build/*,docs/*,mu/contrib*,mu/modes/api/*,utils/*,venv/*,.vscode/* --ignore=E731,E402,W504
find . \( -name _build -o -name var -o -path ./docs -o -path ./mu/contrib -o -path ./utils -o -path ./venv \) -type d -prune -o -name '*.py' -print0 | xargs -0 -r pyflakes
./tests/debugger/test_client.py:579: use ==/!= to compare str, bytes, and int literals
Makefile:51: recipe for target 'pyflakes' failed
make: *** [pyflakes] Error 123
```

My system:
Ubuntu 16.04
Python 3.5.2

I found that I needed to install a few missing dependencies:

```
sudo apt install libffi6 libffi-dev libssl-dev
```

prior to running `pip install -r requirements.txt`

Any help is appreciated!

ntoll, 2019-01-25:
Hmmm... that is strange. I'm on 18.04, Python 3.6.7 and didn't encounter this problem. I wonder if that's just a difference between default packages on Ubuntu.

wayoda, 2019-01-26:
I did a quick test on my Chromebook Linux shell and ran into the same error. Using python 3.5.2 in the virtual environment raises the same error.  
But I did not have to install any addional packages. After editing the line in question `./tests/debugger/test_client.py:579` :
```assert  bp.ignore is 5```
to
```assert bp.ignore == 5```
`make check` completes without error.
The error reported by pyflakes is actually there for a good reason. Because `bp.ignore is 5` does not test for equality of values. It tests if `bp.ignore` and  `5` are the same object. 
see https://wsvincent.com/python-wat-integer-cache/
But like @ntoll `make check` does not trip over this on my desktop machine where I use `mu` in a python 3.6.x environment

ShawnHymel, 2019-01-26:
Got it. Looks like making the same change will let `make check` pass now. Thank you.
I'm working on adding some boards to mu. Would you like me to keep the `bp.ignore == 5` in there when I do a PR?

ntoll, 2019-01-26:
WRT `bp.ignore == 5` in the PR, yes please. Thank you!

ShawnHymel, 2019-01-26:
Great, thank! Closing out this issue.

closed 2019-01-26 by ShawnHymel