mu-editor/mu #756: Context (right-click) menus should be localized.

tmontes, 2019-02-02:
## Environment and Version

* macOS 10.12.6
* Packaged Mu 1.0.2 from https://codewith.mu/en/download.

## Observed behaviour

* Even though Mu shows its interface in the System Language (Portuguese in my testing), the contextual menus in the editor, running, and REPL, panes are displayed in English.

## Expected behaviour

* Contextual menus should be localised as well.

## Steps to reproduce

* Launch Mu such that its UI is in a non-default English language:
  * On macOS, select a different "Preferred Language" under the "Language & Region" section in the "System Preferences" application.
  * On Linux, setting the environment variable `LANG` to something like `de-DE`, `es-ES`, `fr-FR`, or `pt-PT` should work but I didn't test it myself.
  * On Windows, no idea! :(
* Create a new file and right click the editor pane:
  * A popup menu is displayed with the following English entries: "Undo", "Redo", "Cut", "Copy", "Paste", "Delete", "Select all".
* Run that file and right-click the running pane:
  * A popup menu is displayed with the following English entries: "Copy", "Paste".
* Launch the REPL and right click its pane:
  * A popup menu is displayed with the following English entries: Cut, Copy, Copy (Raw Text), Paste, Select All, Save as HTML/XML, Print.

## Wrapping notes

* Could not test on Windows: my system is English and there's no quick way of changing that or Mu.

wayoda, 2019-05-09:
Just checked it. The same on Windows and Linux. 
Looks like main application is not aware it is run in a non-english locale. So standard components like Message-Boxes, the FileDialog and the editor assume english is the preferred language.
The Qt-lib  uses its own system to localize messages, different from the one used in Mu.
So there are 2 ways of dealing with the problem:
1. Turn on the localization used by Qt so the components are aware of a non-english locale and display the translated text for their internal buttons or context menus
2. Provide the translation manually in the code like I proposed in #832 which would require more changes, but gives full control over what is shown to the user.

wayoda, 2019-05-09:
Got this working by loading the default translations for the editor manually from the PyQt library. 

**But** only a few languages are supported by the QScintilla Editor (translation-files shipped with the library), and on my System not all of them work :-( 
But that might be a local problem on my machine. Maybe they work on a different OS?
These are the languages that are in the Qt-Lib
* english (default)
* czech  (doesn't work!)
* portugese (doesn't work! But I don't get an error loading the resource like with french . Anyway Context-menus are not translated)
* spanish (Ok)
* french (doesn't work!)
* german (Ok)

tmontes, 2019-05-11:
Closing. #834 describes the wider Mu/Qt/QScintilla localisation issue, including this issue's particular detail.

closed 2019-05-11 by tmontes