mu-editor/mu #757: Window sizing changes (and the turtle module)

tmontes, 2019-02-02:
## Context

In preparation for an upcoming introductory workshop for kids, mostly based around playing with the [`turtle`](https://docs.python.org/3/library/turtle.html) module in the Standard Library, I've been using/exploring/pushing Mu more and more, recently (thus the bunch of issues I am creating).

I'm using it on macOS and, every now and again, I try to explore/reproduce my findings on Windows 10.

I've read through the excellent [User Experience document](https://mu.readthedocs.io/en/latest/user-experience.html) and took it under consideration in writing these words.

## Mu, the `turtle` module, and beginners

The `turtle` module creates its own "Python Turtle Graphics" window (PTG, from hereon), where the graphics output is displayed. This, while second nature to many, raises the following usability issues:

* Beginner must now track two windows: Mu's, with their code/REPL, and PTG's.
* PTG's window is, unavoidably, created/destroyed every time the program/REPL is started/stopped.
* PTG's window is positioned by the operating system and, on some occasions, at least on macOS, is displayed behind Mu's own window, making the issue even worse.

This becomes seriously distracting/limiting as beginners go into their own edit/run cycles in their learning + trying out new things processes: each time they'll need to locate PTG's window and, probably, size it and position it in such a way that they can see both their code and the graphical outcome.

Faced with this, I figured I would guide them into using an organised side-by-side layout with non-overlapping windows:

* Positioning and sizing Mu's window towards the left half of the screen.
* Giving them a "mini code recipe" to automatically position and size their PTG's window with something like `turtle.Screen().setup(width=600, height=600, startx=740, starty=40)`.

## Current limitation

To my surprise, Mu's window width is currently limited to a minimum of 820 pixels which, I'd say, is more than half the width of many displays. While some (most?) are 1920 pixels wide and more, mine, for example, is 1440, and many others will be less wide (I'd even take a guess that beginners will tend to use simpler / smaller / cheaper computers which tend to have displays with less pixels).

This can severely limit learning scenarios using the Mu window on the left, and PTG's on the right, side-by-side, non-overlapping.

## Proposal

While I'm not sure where the 820 pixel minimum window width comes from, I propose such a limit to be changed to half the screen width.

Pros:
* Support the side-by-side non-overlapping window layout I've been describing.
* Very much inline with common window layouts used in both Windows and macOS (example: on Windows, hitting WINDOWS_KEY+LEFT makes the current window occupy the full left size of the display).

Cons:
* None that I could identify.

Related thoughts:
* Keeping a minimum window size for Mu is still a good usability feature.
* For consistency, I'd limit the window height to half the screen height, instead of the current 400, which, much like the minimum width limit, feels somewhat arbitrary (not sure if that is the case, though).
* Half-screen wide Mu will probably be wide enough for beginner level Python code which will have very little, if any, indentation levels (depends, of course, on both screen width and zoom level).
* Multi-display complexities should be completely ignored as long as Mu does not "explode like crazy" in such scenarios: beginners won't be using multiple displays.

## Going further: remembering window size and position

Given the case I've been describing, mitigated by working with a side-by-side Mu + PTG window layout, having Mu relaunch with the same window size and position it had when it quit previously could be very nice for beginners.

AFAICT, many programs work that way. Others don't. Being mostly used to macOS, I see that most GUI programs behave that way, remembering their window position and size. Not so sure about the other desktop platforms.

Pros:
* Zero time wasted sizing / positioning the Mu window when following up with work that depends/works along with "third party" windows, like the one from the `turtle` module.

Cons:
* Could not identify any.

## Parting thoughts

* I have both these ideas implemented in different branches:
  * [Save/restore window position and size](https://github.com/tmontes/mu/tree/save-window-state).
  * [Allow smaller window](https://github.com/tmontes/mu/tree/allow-smaller-window).
* If this proposal is accepted, I'm willing to work further on this until such branches are mergeable (in particular, they're lacking tests). :)

PS: Going completely crazy would be having the PTG window "swallowed in" as a Mu window pane, towards the right size, much like the debugger. That would be a non-trivial endeavour though: we would need to i) override the running program's tkinter module, ii) proxy its calls back to Mu via a pipe/socket or some other IPC mechanism, iii) map proxied calls to Qt primitives. Crazy, but nice! :)

ntoll, 2019-02-19:
Thanks for this investigation. The window width is based upon the size needed for the Python3 mode to display without any of the buttons becoming "truncated". If the width of the window is less than the width needed to display the buttons, then the Quit button (and others) are accessed by a sort of "open more" band which you need to click. When testing Mu, this wasn't particularly obvious for novices, so I enforced the minimum width for the window to mitigate. Does this make sense?

I like the idea of save/restore window position and size -- this is often asked for.

tmontes, 2019-02-19:
> Thanks for this investigation. The window width is based upon the size needed for the Python3 mode to display without any of the buttons becoming "truncated". If the width of the window is less than [...]

It does indeed. That's why I consider this topic intimately related to #758 where I raise the button scaling adjustments issue. I think, however, that the priority of topics should be:

* Allow window size to be narrow enough for side-by-side things (like the Turtle window, or a Pygame Zero window): half screen width, while somewhat arbitrary, means users can divide their screen space equally between their code in the Mu window, and its outcome in whichever other window. :) The default size/position, centred with 80% screen width/height feels perfectly good.

* Following that, scale the icons automatically such that they're always visible (within reason, of course: I'm not saying we should smaller that the current small 32x32px sizes)

* Challenge: different UI translations do lead to different button spacing... *sigh!* :)

tmontes, 2019-07-04:
Closing after the merge of #775. Intimately related, IMHO, is reviewing #758.

closed 2019-07-04 by tmontes