mu-editor/mu #760: File Name Tab resizing

Farshidmi, 2019-02-04:
The appearance and disappearance of the * will change the **size** of the File Name TAB and It seems to be the Students distraction when we have two files.

 It seems that something is blinking on screen when autosave happen.
 I suggest TAB size does not change while adding or removing *  . 
Reagrds
 
[image]

ZanderBrown, 2019-07-16:
Hmm, wonder if there is a less disruptive way to indicate a dirty buffer

Theoretically, we could change border colour or something

tmontes, 2019-07-17:
Went looking at VSCode's behaviour which we might adopt that might both be better in terms of accessibility (not colour dependent, for example) and simple to implement (guessing, here!): when the buffer is unsaved, the "x" closing mini-button becomes a filled "o". Saving turns the filled "o" back into the traditional "x" close button.

Could this be interesting? :)

ZanderBrown, 2019-07-17:
Didn't think about code (ironic given how much of my day is staring into it), not sure changing the "X" is the most "intuitive" approach though

As a colour deficient person it's nice to know someone is thinking about the downsides of representing state though colour :-)

carlosperate, 2019-07-17:
> when the buffer is unsaved, the "x" closing mini-button becomes a filled "o". Saving turns the filled "o" back into the traditional "x" close button.

To be fair I found that slightly confusing the first time I encountered that pattern (possibly in VSCode). I'm used to it now, but I wonder how common is it in general?
I don't know, removing the close icon feels a bit unintuitive to me, but if is a know pattern it shoudn't be a problem.

I think the `*` works well, but the original report is right in that changing the tab width is quite distracting, if the font used is monospaced we could just add a blank character?

ZanderBrown, 2019-07-17:
> font used is monospaced we could just add a blank character

Well there is always that trick :-)

We should probably be using • instead of * anyway

/me heads off to Qt docs

ZanderBrown, 2019-07-17:
How about

[image]

Where the first tab is dirty, notice the dot changes the outline thus isn't dependent on colour

Window title is still suffixed as before

tmontes, 2019-07-17:
@ZanderBrown, that could work. I never gave it much thought, but I kind of like VSCode's approach: the "x" becomes a filled "o", which when mouse-hovered becomes an "x" which feels like a clean, discreet enough / in-your-face enough way of telling the user "you can close it, but it is not saved".

More ideas? :)

ZanderBrown, 2019-07-17:
@tmontes having spent a while in the Qt docs I don't think that's doable, if this was Gtk it would be easy

Even if it were doable I'm not sure it makes much sense, tabs actually backed by a file are autosaved and those which aren't throw a warning on close

tmontes, 2019-07-17:
> Even if it were doable I'm not sure it makes much sense, tabs actually backed by a file are autosaved and those which aren't throw a warning on close

...maybe it doesn't! :) I just wonder if the "extra icon on the left" solution is too noisy, visually. Hmmm, thinking how such an apparently "trivial" issue, may need lots of thinking, usability-wise, util a good solution is found. :)

ZanderBrown, 2019-07-17:
> solution is too noisy, visually

Yh that concerned me as well

There is nothing stoping the icon for a clean state being empty space

tmontes, 2019-07-17:
> There is nothing stoping the icon for a clean state being empty space

That might actually be a good idea / hack! :)

closed 2019-09-11 by ntoll