mu-editor/mu #762: UI/UX | Check button vs. Positive feedback

tmontes, 2019-02-04:
## Context

I've been using Mu recently, more and more, while preparing an introductory programming workshop for kids. One of the things I've been trying really hard is to "step into their experience level" as much as possible. From that, I have found a few UI/UX issues I think can be improved.

## The issue

Much like the "Mode button vs. Current mode indicator" issue #761 I just logged, the "all ok" feedback and the Check button that triggers it are on completely opposite "locations" within Mu's window:

[image]

While the "not ok" feedback is displayed inline with the code, which I find good, the "all ok" friendly feedback can easily be missed (possible beginner's experience: hmmm, let me click this button here to "check".... nothing happened?... maybe I'm not understanding... moves on with a an feeling of "I didn't get that one!").

## Proposal (for discussion, of course!)

Let's have the "all ok" positive and friendly feedback be displayed more "in-your-face" as a reaction to clicking the Check button. Here's a possible idea where a tooltip like message would be displayed right under the button:

[image]

Notes:
* I went for a somewhat larger font size, given that "all ok" messages tend to be short sentences, helping the differentiation from tooltips which are displayed in a smaller font and a grayish background.
* A yellow background (same yellow as the button borders') helps the feedback stand out in the white/grayish background.
* Not to be implemented as a tooltip: the message would only go away once any mouse click and/or keyboard typing happens anywhere within Mu. Not timeout based.
* A variation on the idea: overlay a semi-transparent grey layer on top of the editor and display the "all ok" message in a vertically and horizontally centered yellow background box, pretty much like the one above. This would be a more in-your-face feedback and behave exactly the same, going away after a click.
* In any case, the status bar feedback would be completely eliminated.

## Related thoughts

Given that the check button is somehow stateful, acting as a "highlight/hide problems" toggle when there are "problems" in the code, maybe the overall behaviour could be improved in a way that its own icon could change depending on the current state:

* After "all ok" feedback, it would become the current "thumbs up" icon.
* After an "errors found" feedback, it could become a "thumbs down" icon variation.
* Any editing or front tab selection change would reset the button an "idle, i don't know" icon.
* The "idle, i don't know" icon would need some thought... Here are a few ideas I played around with:

| Idle, I don't know | All ok | Problems found |
| ----------------- | ------- | --------------- |
| [image] | [image] |  [image] |
| [image] | | | 

One last thought:
* Does the button position, towards the right hand side of the button bar, make sense? Would it be better placed if it were immediately to the left of the "run" / "play" / "flash" button? I'm not sure at all and would love to hear other people's thoughts on that.

## Epilogue

As before, the fundamental idea is to improve the experience in the "all ok" feedback case. Other ideas are related but not fundamental.

Your words are welcome.

PS: With barely any Qt experience, I have no idea how to implement what I'm proposing here! :)

ntoll, 2019-02-19:
Thank you for this! Some feedback and explanation.

The [opening tutorial on Mu](https://codewith.mu/en/tutorials/1.0/start) explains the reasoning behind the button grouping and placement.

Please see the discussion in #761 about the three tier (buttons, code, status) layout we use with Mu.

When I first implemented this I did experiment with "thumbs up" / "thumbs down" icons to indicate state but had negative feedback from users (it wasn't clear how the two images related to each other, the user's code nor what they actually meant). Far better to have a single image that merely associated with checking if the code was OK (hence the choice of the thumbs up "OK" image).

I wonder if we should implement some sort of call to the OS notification service on Windows, OSX and Linux, so important notices are displayed in the message queue? Thoughts..?

More UX discussion can be found in the developer docs here: https://mu.readthedocs.io/en/latest/user-experience.html

tmontes, 2019-02-19:
Nicholas,

Thanks for going over yet another proposal. :)

* I tend to agree on the button thing (that's why it was presented as a related thought): your feedback confirms that, good. Let's keep the button as is.

* The thing I think could be improved is the positive feedback solution:

  * The kind of tooltip I shared is more visible, but inconsistent with everything else.

  * Using OS notifications may lead to a somewhat awkward experience (???) and, importantly, to a development can of worms I'm not sure is easy to tame and maintain (unless Qt solves that for us).

  * Given that the left side of status bar is used for several informative messages, maybe it could somehow try to grab the user's attention on some occasions such as this one, reacting to a button click and saying "your code looks good"? How? Sliding the message in? Flashing the message somehow?

  * Not sure, really... :)

PS: Thanks for sharing the link to the developer docs (which I did read carefully before creating this set of UI/UX related issues).

tmontes, 2019-02-21:
One tidbit on OS notifications:

* Qt seems to handle that. I just tried this bare bones script successfully, in the REPL, on macOS:
```
from PyQt5.QtWidgets import QSystemTrayIcon, QApplication
app = QApplication([])
sti = QSystemTrayIcon()
sti.show()
sti.showMessage("Hello", "There", 0)
sti.hide()
```
* However, both [Apple's](https://developer.apple.com/design/human-interface-guidelines/macos/system-capabilities/notifications/) and [Gnome's](https://developer.gnome.org/hig/stable/notifications.html.en) docs point towards the idea that notifications are to be used by applications not currently in the foreground:

  * Apple says: "**Handle notifications gracefully if your app is in the foreground**. (...) Present it to the user in a way that’s informative but not distracting or invasive. For example, you might display or increment a badge or subtly insert new data into the current view.".

  * Gnome says: "**When to use** (...) Notifications shouldn’t be used as a substitute for feedback that is provided by your application windows, which should be able to inform the user about events without the need for notifications."

ntoll, 2019-02-21:
@tmontes oooh... nice find. I hadn't thought of Qt (I looked at OS specific notification Python libs).

However, the UI guidelines are, upon reflection, probably to be taken into account.

tmontes, 2019-02-22:
> @tmontes oooh... nice find. I hadn't thought of Qt (I looked at OS specific notification Python libs).

Not being very familiar it, I also keep forgetting that Qt is much more than a cross-platform GUI toolkit: it is a cross-platform SDK! Thus the locale detection simplification code in #764. :)

> However, the UI guidelines are, upon reflection, probably to be taken into account.

Agreed.

tjguk, 2019-02-22:
Regardless of its widespread use, I've never found the "status bar message" to be useful for information which should catch your eye. It's useful for information which is essentially static or ongoing, including current filename, line/col of cursor, saved/not saved etc. Where the eye's focus will be on a button which is nowhere near the status bar, having the result of pressing that button appear there is unlikely to attract attention.

The "toaster" pop-ups are a bit of an attractive nuisance in general. Even if -- per the UI guidelines -- they only activate when the app is inactive. And especially when it's active as the suggestion is here.

I have a halfway suggestion but we'd have to tune it carefully: we could, as part of the autosave or whenever, run the Check process in the background. We don't display the check hints as now, but we do colour the button (or otherwise tweak the appearance). This gives an up-front indication of whether there's might be a problem, at which point the user can choose to press [Check] or not. That information *could* be in the status bar because it's essentially ongoing information which you can action.

To take it a little further, we could show one visual indication for syntax errors which means the code wouldn't work, and another for layout / style although we're then treading on Black territory.

ntoll, 2019-09-11:
OK folks, so @ZanderBrown's rather cute animation red/green effect on the button seems to address much of the discussion here without the negative side effects (annoying toaster popups via the OS).

I'm going to merge it so we can present it to users in the next alpha so they can tell us either how much they love it or hate it.

closed 2019-09-11 by ntoll

tjguk, 2019-09-11:
Amusingly, I have absolutely no memory of writing whatever I appear to have written above. However it does seem to be sensible feedback so I'm going to stand by it anyway...

ntoll, 2019-09-11:
@tjguk I know exactly what you mean. :-)