mu-editor/mu #763: Much simpler locale detection code

tmontes, 2019-02-04:
While investigating / preparing a UI proposal to address #518 and #692, I learned that given that Mu depends on Qt, there's a pretty direct way of handling what all the code in merged #692 does, with two lines of code.

Tested on macOS 10.12.6, with the UI language as English:

```python
>>> from PyQt5.Qt import QLocale
>>> QLocale.system().name()
'en_US'
```

...when the UI language is set to Portuguese it returns `pt_PT`, and when set to Spanish, returns `es_ES`.

I do believe that its behaviour will be equally correct on the other platforms, but it does need confirmation.

What do you say of throwing away 1 module with 80 lines of code?

carlosperate, 2019-02-06:
Does it work well in macOS when the app is launched from finder instead of the terminal (in which case #672 had to run `defaults read -g AppleLocale`)?

tmontes, 2019-02-06:
A quick test, after hand hacking the `mu-editor.app/Contents/Resources/app/mu/localedetect.py`file down to...

```
from PyQt5.Qt import QLocale

def language_code():
    return QLocale.system().name()
```

...shows that Finder launching works and correctly detects the user preference in the "Language & Region" pane in the "System Preferences".

Note, however, that the key motivation for #672 wan't Finder vs. terminal launching but, instead, following the macOS user preferences I mentioned above which do not alter language/locale related environment variables (set by the terminal, but not by Finder when launching processes). :)

tmontes, 2019-02-18:
#764 has been merged, this issue can now be closed.

closed 2019-02-18 by tmontes