mu-editor/mu #765: chdir to code before running?

tjguk, 2019-02-07:
Thanks to #726 we can now import other modules in the same directory when Mu is running code outside mu_code. However... the cwd is still the mu_code directory. So any data files need to be fully qualified. (I discovered this the hard way when trying to do a quick and dirty demo of loading and saving data which became slower and dirtier!)

It occurs to me that a simple solution is to chdir to the directory in which the code file sits before running (ie set the cwd of the subprocess to that directory). On the surface this would seem to be an easy fix and I can't see any downsides.

I'm happy to produce a PR

tjguk, 2019-02-07:
Although I hope that the situation is obvious, here's an example file which I saved in `c:\temp\mutest` with a file called `a.txt` in the same directory.

```
import os, sys
print("CWD:", os.getcwd())
print("sys.path:")
for p in sys.path:
    print("  ", p)
with open("a.txt") as f:
    print(f.read())
```

and here's the output from within Mu:

```
CWD: c:\users\goldent\mu_code
sys.path:
   c:\temp\mutest
   c:\users\goldent\mu_code
   C:\Users\goldent\.venvs\mu\Scripts\python36.zip
   C:\Program Files\Python36\DLLs
   C:\Program Files\Python36\lib
   C:\Program Files\Python36
   C:\Users\goldent\.venvs\mu
   C:\Users\goldent\.venvs\mu\lib\site-packages
   C:\Users\goldent\AppData\Local\python\mu\site-packages
   C:\Users\goldent\.venvs\mu\lib\site-packages\IPython\extensions
Traceback (most recent call last):
  File "c:\temp\mutest\a.py", line 6, in <module>
    with open("a.txt") as f:
FileNotFoundError: [Errno 2] No such file or directory: 'a.txt'
```

carlosperate, 2019-02-07:
This is related to the second part of https://github.com/mu-editor/mu/issues/480.

I thought this was the updated behaviour, as PR  #739 closed the issue, but this PR only updates the file save dialog default location.
Based on that conversation I got the impression that it was agreed that the cwd should be changed to the one from the python file (so files are opened there) when running/debugging the files. Not 100% sure anymore after a second read.

From my point of view we should do this, the "default location for files" requirement is already satisfied by saving new files into mu_code by default, and a user running a file in any other directory would expect the relative paths to be all from that file.

tjguk, 2019-02-07:
Agree with @carlosperate assessment -- I'd overlooked that element of #480.

Knowing that @ntoll is travelling, I'll wait for him to respond but AFAICT the change is fairly trivial, although we need to do it on a mode-by-mode basis.

tmontes, 2019-02-07:
### Quick analysis and comparison

Take PFD to mean "Python File's directory", other than Mu's default `~/mu_code`.

These were collected with a script very similar to the one @tjguk shared above, but also including an `import` of a module to be sourced from a "sibling" Python file in the PFD.

|    | Mu | IDLE | Thonny | VS Code |
| -- | -- | ---- | ------ | ------- |
| `os.getcwd()` | `~/mu_code` | PFD | PFD | PFD |
| `open` files in PFD | needs path | filename only ok |  filename only ok |  filename only ok |
| `sys.path` |  PFD is there | PFD is there | `''` is there | PFD is there |
| `import` from PFD | works | works | works | works |

Observations:
* Other IDEs do `os.chdir` to the PFD.
* This makes life easier (and maybe more natural) to many, especially beginners.
* Today's Mu users may grow into other tools like these, which seem to behave according to this proposal -- consistency is good.

Side note, for "full coverage":
* This is not inline with advanced CLI usage where running `python3 path/to/script.py` does not change the CWD, but adds the PSD to `sys.path`, pretty much like Mu is doing today.
* Such advanced CLI usage is way out of scope, my Mu's own purposes. :)

### Conclusion

Sure, Mu's runner/debugger should `os.chdir` to the directory of the Python file to be run/debugged.

ntoll, 2019-02-07:
OK... finally found some Wifi.

@tjguk @tmontes @carlosperate thank you for looking / responding to this. I'm about to head off to give my presentation at PyData Edinburgh, so briefly, I'm +1 on this being a problem and the suggested fix (`os.chdir` to the file being run/debugged).

tjguk, 2019-02-07:
Un-formally-tested PR available if anyone wants to take it for a spin. Not sure we need to / should do anything for the micro:bit/CircuitPython modes?

tjguk, 2019-03-30:
This was merged in back in Feb

closed 2019-03-30 by tjguk