mu-editor/mu #767: Mu 1.0.2 not saving files to Circuit Playground Express on a Mac

tdm47, 2019-02-09:
If you are reporting a bug, we would like to know:

* What you were trying to do,
* What steps you took to make this happen,
* What you expected to happen,
* What actually happened,
* Why this difference is problematic (it may not be a bug!),
* Technical details like the version of Mu you're using, your OS version and
  other aspects of the context in which Mu was running. 

Please remember to attach a **copy of the full log files for Mu**. You can get
the logs by clicking on the cog icon in the bottom right of the editor window.
Click on the logs and use CTRL-A to select all, then CTRL-C to copy and CTRL-V
to paste the contents into the issue.

Thank you for contributing to Mu! :-)

tdm47, 2019-02-09:
I'm using a Mac with Mac OS 10.13.6 and Mu Editor 1.0.2 and a new Circuit Playground Express. Everything was working well using MakeCode and Circuit Python then the "save" function for Circuit Python ceased to work. A test code from the REPL got a response from the Playground express and the system worked ok back on MakeCode. The Mu file follows.`2019-02-10 06:32:05,773 - root:111(run) INFO: 

-----------------

Starting Mu 1.0.2
2019-02-10 06:32:05,774 - root:112(run) INFO: uname_result(system='Darwin', node='Ruths-MBP.gateway', release='17.7.0', version='Darwin Kernel Version 17.7.0: Wed Oct 10 23:06:14 PDT 2018; root:xnu-4570.71.13~1/RELEASE_X86_64', machine='x86_64', processor='i386')
2019-02-10 06:32:05,774 - root:113(run) INFO: Python path: ['/Applications/mu-editor.app/Contents/MacOS', '/Applications/mu-editor.app/Contents/Resources/python/lib/python36.zip', '/Applications/mu-editor.app/Contents/Resources/python/lib/python3.6', '/Applications/mu-editor.app/Contents/Resources/python/lib/python3.6/lib-dynload', '/Applications/mu-editor.app/Contents/Resources/python/lib/python3.6/site-packages', '/Applications/mu-editor.app/Contents/Resources/app', '/Applications/mu-editor.app/Contents/Resources/app_packages', '/Applications/mu-editor.app/Contents/Resources/app_packages/IPython/extensions']
2019-02-10 06:32:05,774 - root:114(run) INFO: Language code: en_AU
2019-02-10 06:32:05,981 - mu.logic:543(__init__) INFO: Setting up editor.
2019-02-10 06:32:05,982 - mu.logic:562(__init__) INFO: Settings path: /Users/ruthbest/Library/Application Support/mu/settings.json
2019-02-10 06:32:05,982 - mu.logic:563(__init__) INFO: Session path: /Users/ruthbest/Library/Application Support/mu/session.json
2019-02-10 06:32:05,982 - mu.logic:564(__init__) INFO: Log directory: /Users/ruthbest/Library/Logs/mu
2019-02-10 06:32:05,982 - mu.logic:565(__init__) INFO: Data directory: /Users/ruthbest/Library/Application Support/mu
2019-02-10 06:32:05,994 - mu.logic:578(setup) INFO: Available modes: python, adafruit, microbit, debugger, pygamezero
2019-02-10 06:32:06,112 - mu.logic:1150(change_mode) INFO: Workspace directory: /Users/ruthbest/mu_code
2019-02-10 06:32:06,122 - mu.logic:627(restore_session) INFO: Restoring session from: /Users/ruthbest/Library/Application Support/mu/session.json
2019-02-10 06:32:06,122 - mu.logic:628(restore_session) DEBUG: {'theme': 'night', 'mode': 'adafruit', 'paths': ['/Users/ruthbest/mu_code/test.py'], 'envars': [], 'minify': False, 'microbit_runtime': '', 'zoom_level': 2}
2019-02-10 06:32:06,123 - mu.logic:715(_load) INFO: Loading script from: /Users/ruthbest/mu_code/test.py
2019-02-10 06:32:06,123 - mu.logic:274(read_and_decode) DEBUG: Trying to decode with utf-8
2019-02-10 06:32:06,123 - mu.logic:277(read_and_decode) INFO: Decoded with utf-8
2019-02-10 06:32:06,124 - mu.logic:290(read_and_decode) DEBUG: Detected newline '\n'
2019-02-10 06:32:06,124 - mu.logic:789(_load) DEBUG: import board
import digitalio
import time

led = digitalio.DigitalInOut(board.D13)
led.direction = digitalio.Direction.OUTPUT

while True:
    print("Hello, CircuitPython!")
    led.value = True
    time.sleep(1)
    led.value = False
    time.sleep(1)
2019-02-10 06:32:06,175 - mu.logic:649(restore_session) INFO: Loaded files.
2019-02-10 06:32:06,175 - mu.logic:653(restore_session) INFO: User defined environment variables: []
2019-02-10 06:32:06,175 - mu.logic:657(restore_session) INFO: Minify scripts on micro:bit? False
2019-02-10 06:32:06,196 - mu.logic:1150(change_mode) INFO: Workspace directory: /Volumes/CIRCUITPY
2019-02-10 06:32:18,521 - mu.modes.base:227(find_device) INFO: Found device on port: cu.usbmodem1421
2019-02-10 06:32:18,521 - mu.modes.base:228(find_device) INFO: Serial number: B58A1F9B3323B4050213E273319110FF
2019-02-10 06:32:18,541 - mu.modes.base:278(add_repl) INFO: Started REPL on port: /dev/cu.usbmodem1421
2019-02-10 06:32:18,541 - mu.modes.base:259(toggle_repl) INFO: Toggle REPL on.
2019-02-10 06:32:22,007 - mu.logic:871(save_tab_to_file) INFO: Saving script to: /Users/ruthbest/mu_code/test.py
2019-02-10 06:32:22,007 - mu.logic:872(save_tab_to_file) DEBUG: import board
import digitalio
import time

led = digitalio.DigitalInOut(board.D13)
led.direction = digitalio.Direction.OUTPUT

while True:
    print("Hello, CircuitPython!")
    led.value = True
    time.sleep(1)
    led.value = False
    time.sleep(1)
2019-02-10 06:33:52,605 - mu.logic:871(save_tab_to_file) INFO: Saving script to: /Users/ruthbest/mu_code/test.py
2019-02-10 06:33:52,608 - mu.logic:872(save_tab_to_file) DEBUG: import board
import digitalio
import time

led = digitalio.DigitalInOut(board.D13)
led.direction = digitalio.Direction.OUTPUT

while True:
    print("Hello, CircuitPython!")
    led.value = True
    time.sleep(1)
    led.value = False
    time.sleep(1)
2019-02-10 06:34:02,157 - mu.logic:1077(show_admin) INFO: Showing logs from /Users/ruthbest/Library/Logs/mu/mu.log
2019-02-10 06:51:10,179 - mu.logic:1077(show_admin) INFO: Showing logs from /Users/ruthbest/Library/Logs/mu/mu.log
`     Thank you for any suggestions.

tdm47, 2019-02-11:
The code is now posted (with format) in the MuEditos/General Gitter page

carlosperate, 2020-11-08:
@tdm47 were you able to resolve this issue?

I am not too sure how the circuit playground works, but it is possible that if you were using MakeCode before you might have to flash CircuitPython in "bootloader mode", before you are able to flash a Python file into the CIRCUITPY drive.

carlosperate, 2021-03-24:
@tdm47 were you able to resolve this issue?

carlosperate, 2022-07-20:
Close due to lack of activity.  If this was an issue with app permissions to the USB drive it should be resolved in the latest releases.