mu-editor/mu #768: "Include" doesn't work with local python files

zlite, 2019-02-11:
Use of "include file" where "file" is a filename.py in the same directory as the parent script fails with a "module not found" error. Meanwhile the same combination works fine in Idle

Tested on Python 3.x

ntoll, 2019-02-11:
Not sure what you mean. Care to share some code that demonstrates this please? Perhaps you mean `import`. What version of Mu are you using (see the title bar of Mu's Window).

ntoll, 2019-02-11:
If you *do* mean the issue is with `import` and Python modules in the same directory as the script being run, then this is fixed in Mu version 1.0.2 (please update your version of Mu).

zlite, 2019-02-12:
Yes, sorry -- I did mean "import". (autocorrect!)
Thanks for the update!

closed 2019-02-12 by zlite

ntoll, 2019-02-12:
No problem. Glad it's working (I presume, you've updated to 1.0.2).