mu-editor/mu #769: When saving files on mac, final newline isn't saved

makermelissa, 2019-02-12:
If you are reporting a bug, we would like to know:

* What you were trying to do,
Edit files
* What steps you took to make this happen,
On a mac (haven't checked windows), do the following:
   1. Create a new file
   2. Make sure it is 10 lines long with a couple empty lines at the beginning
   3. Save the File
   4. Close the tab
   5. Open the file
   6. Notice there is 1 less line than before
* What you expected to happen,
   That there would be the same number of lines
* What actually happened,
   The last newline character was stripped
* Why this difference is problematic (it may not be a bug!),
   When trying to create CircuitPython libraries, then needs to be one empty line and this frequently results in having to commit code an extra time to fix it.
* Technical details like the version of Mu you're using, your OS version and
  other aspects of the context in which Mu was running. 
Mu 1.0.2 and MacOS 10.14.3

Please remember to attach a **copy of the full log files for Mu**. You can get
the logs by clicking on the cog icon in the bottom right of the editor window.
Click on the logs and use CTRL-A to select all, then CTRL-C to copy and CTRL-V
to paste the contents into the issue.

Thank you for contributing to Mu! :-)

tjguk, 2019-02-12:
Thanks for reporting this, @makermelissa. That was actually a semi-conscious decision -- ie I was aware that any trailing newline would be stripped. It was just slightly easier to code and I didn't think it would adversely affect anyone. (Famous Last Words).

I'm not sure I entirely understand the impact on CircuitPython although I'm quite willing to accept your word for it that it matters. At a guess, I think you're saying that you can concatenate libraries together but that there must be a blank line between each. Have I got that right?

Either way, I'm happy to push a fix for this. However... if you fancied producing a PR yourself, I'd happily review it. If you're interested and needed help, I can guide you through the process.

makermelissa, 2019-02-12:
Hi @tjguk, the impact is that when CircuitPython files are pushed up to github, Travis-CI checks them for a newline character. Sometimes I'll grab a file to do a really quick edit in the middle of the file that passes the checks just fine, but mu stripped off the final newline character and I need to go to the bottom of the file, add an extra newline and recommit again. It's just an extra unnecessary step. I forked it and I'll go ahead and see what I can find. Thanks.

PR #771 (2019-02-20): Adds stripped newline to end
I'm not sure if this is the best way to go about it, but it fixes #769. The other behavior (which I think is desirable) is that if the file does not have a newline at the end, this adds it. I edited the .py file in my Mu Editor App folder and tested that this change works.
closed 2019-02-20 by ntoll