mu-editor/mu #774: Crashes every couple days on Mac

makermelissa, 2019-02-15:
If you are reporting a bug, we would like to know:

* What you were trying to do,
Open a File
* What steps you took to make this happen,
Haven't been able to determine steps to replicate this
* What you expected to happen,
For it not to crash
* What actually happened,
Mu crashed
* Why this difference is problematic (it may not be a bug!),
Potential loss of work
* Technical details like the version of Mu you're using, your OS version and
  other aspects of the context in which Mu was running. 
Mu 1.0.2

Here's the relevant log entries related to the crash event:
```
2019-02-15 10:18:47,066 - mu.modes.base:256(toggle_repl) INFO: Toggle REPL off.
2019-02-15 10:18:47,545 - mu.modes.base:227(find_device) INFO: Found device on port: cu.usbmodem14101
2019-02-15 10:18:47,546 - mu.modes.base:228(find_device) INFO: Serial number: 372360C41F414397
2019-02-15 10:18:47,560 - mu.modes.base:278(add_repl) INFO: Started REPL on port: /dev/cu.usbmodem14101
2019-02-15 10:18:47,560 - mu.modes.base:259(toggle_repl) INFO: Toggle REPL on.
2019-02-15 10:26:32,109 - mu.logic:811(get_dialog_directory) INFO: Using path for file dialog: /Users/melissa/Developer/CircuitPython/Adafruit_CircuitPython_ESP32SPI/adafruit_esp32spi
2019-02-15 10:26:44,951 - mu.interface.main:251(get_load_path) DEBUG: Getting load path: /Volumes/CIRCUITPY/featherwing_sevensegment_simpletest.py
2019-02-15 10:26:44,951 - mu.logic:715(_load) INFO: Loading script from: /Volumes/CIRCUITPY/featherwing_sevensegment_simpletest.py
2019-02-15 10:26:44,951 - root:95(excepthook) ERROR: Unrecoverable error
Traceback (most recent call last):
  File "/Applications/mu-editor.app/Contents/Resources/app/mu/logic.py", line 831, in load
    self._load(path)
  File "/Applications/mu-editor.app/Contents/Resources/app/mu/logic.py", line 730, in _load
    if os.path.samefile(path, widget.path):
  File "/Users/travis/build/mu-editor/mu_portable_python_macos/python/lib/python3.6/genericpath.py", line 97, in samefile
FileNotFoundError: [Errno 2] No such file or directory: '/Volumes/CIRCUITPY/code.py'
2019-02-15 10:27:03,989 - root:112(run) INFO: 
```

The line that concerns me is the one with `/Users/travis` as that path does not exist on my system.

ntoll, 2019-02-15:
Hi @makermelissa,

So the error is telling us that it's the path `/Volumes/CIRCUITPY/code.py` that doesn't exist (i.e. I presume you've unplugged the device)..?

In any case, it's a bug since Mu should be able to cope with you unplugging a device.

makermelissa, 2019-02-15:
Actually, no. I didn't unplug it. I just tried opening a file.

ntoll, 2019-02-15:
OK... so this is the OS reporting to Python that the file doesn't exist. Perhaps the device was plugged in but just not mounted onto the filesystem..?

makermelissa, 2019-02-15:
Oh right. Right before it crashed, I was testing a bug in CircuitPython that causes the nrf52840 to reboot. That's probably what that's from.

makermelissa, 2019-02-15:
But it didn't crash until I tried opening another file.

ntoll, 2019-02-15:
Riiight... so there's two things going on here:

* You're doing something which means the OS can't find the file on the attached device (don't do that!!!!!) :-)
* In any case, Mu should recover from this in a graceful fashion... i.e. instead of crashing, it should pop up a modal dialog and say something helpful/useful. :-)

In any case, you've found a bug (well done) or, at the very least, a cool enhancement for a use case we'd not thought of before. Nice one..! :-D

makermelissa, 2019-02-18:
Had another crash. Here's the log entry:
```
2019-02-17 17:49:47,660 - mu.modes.base:256(toggle_repl) INFO: Toggle REPL off.
2019-02-17 17:49:48,178 - mu.modes.base:227(find_device) INFO: Found device on port: cu.usbmodem14101
2019-02-17 17:49:48,178 - mu.modes.base:228(find_device) INFO: Serial number: 646071AB2345733502020293B324F0FF
2019-02-17 17:49:48,188 - mu.modes.base:278(add_repl) INFO: Started REPL on port: /dev/cu.usbmodem14101
2019-02-17 17:49:48,188 - mu.modes.base:259(toggle_repl) INFO: Toggle REPL on.
2019-02-17 19:57:38,611 - mu.modes.base:256(toggle_repl) INFO: Toggle REPL off.
2019-02-17 19:57:39,110 - mu.modes.base:227(find_device) INFO: Found device on port: cu.usbmodem14101
2019-02-17 19:57:39,110 - mu.modes.base:228(find_device) INFO: Serial number: 372360C41F414397
2019-02-17 19:57:39,124 - mu.modes.base:278(add_repl) INFO: Started REPL on port: /dev/cu.usbmodem14101
2019-02-17 19:57:39,124 - mu.modes.base:259(toggle_repl) INFO: Toggle REPL on.
2019-02-17 20:12:28,741 - mu.modes.base:256(toggle_repl) INFO: Toggle REPL off.
2019-02-17 20:12:29,698 - mu.modes.base:227(find_device) INFO: Found device on port: cu.usbmodem14101
2019-02-17 20:12:29,698 - mu.modes.base:228(find_device) INFO: Serial number: B0BA68C9A46405150202029343A131FF
2019-02-17 20:12:29,711 - mu.modes.base:278(add_repl) INFO: Started REPL on port: /dev/cu.usbmodem14101
2019-02-17 20:12:29,711 - mu.modes.base:259(toggle_repl) INFO: Toggle REPL on.
2019-02-17 20:18:08,130 - mu.logic:811(get_dialog_directory) INFO: Using path for file dialog: /Users/melissa/Developer/CircuitPython/Adafruit_CircuitPython_FeatherWing/adafruit_featherwing
2019-02-17 20:18:13,735 - mu.interface.main:251(get_load_path) DEBUG: Getting load path: /Volumes/CIRCUITPY/main.py
2019-02-17 20:18:13,735 - mu.logic:715(_load) INFO: Loading script from: /Volumes/CIRCUITPY/main.py
2019-02-17 20:18:13,741 - root:95(excepthook) ERROR: Unrecoverable error
Traceback (most recent call last):
  File "/Applications/mu-editor.app/Contents/Resources/app/mu/logic.py", line 831, in load
    self._load(path)
  File "/Applications/mu-editor.app/Contents/Resources/app/mu/logic.py", line 730, in _load
    if os.path.samefile(path, widget.path):
  File "/Users/travis/build/mu-editor/mu_portable_python_macos/python/lib/python3.6/genericpath.py", line 97, in samefile
FileNotFoundError: [Errno 2] No such file or directory: '/Volumes/CIRCUITPY/lib/adafruit_featherwing/rtc_featherwing.py'
2019-02-17 20:18:35,524 - root:112(run) INFO: 
```

I think I figured out what it is that I'm doing. I switch between different devices and leave the file open anyways. When trying to open a new file, it appears to check that other open files exist and if they don't, it crashes.

ntoll, 2019-02-18:
Yup... that sounds like a good assessment of what's going on.

In any case, it's definitely a bug, and I hope to create a fix today or tomorrow..! It'll be fixed in the alpha 1 release coming soon! :-)

ntoll, 2019-02-20:
@makermelissa OK... I just landed a fix for this in `master`. Please confirm it solves your problem (otherwise please feel free to re-open this ticket).

Thanks for the report! :-)

closed 2019-02-20 by ntoll

makermelissa, 2019-02-21:
Ok cool. I just pulled the code and copied it into the App so I'll watch to see if it happens again. Thanks!

commit 2e9fbc16 (2019-02-20): Fixes #774. Ensures that open tabs whose files no longer exist doesn't cause a crash when checking for duplicates.