mu-editor/mu #780: Request mu to allow user to customize which syntax check errors should be ignored

leroyle, 2019-03-01:
This issue/request is exactly like issue #253. I'm hoping the solution that was implemented for that issue can be enhanced to allow for a better user experience.

There was a change to the syntax checking as a result of issue #253:  "Error checking in mu - request #253  "

That issue/request asked for a way to loosen the syntax checking that mu is running so that young kids do not get bogged down in the nuances of coding style that the PEP8 check is enforcing. As mentioned in that issue, coding style is good but does make it confusing for the younger beginners.

To support issue #253 a change was made to logic.py - check_pycodestyle(code) function. That change overrides "any" ignore rules that may have been specified when pycodestyle's StyleGuide() function was called.

The pycodestyle module has a builtin mechanism to allow the user to configure certain aspects of its behavior. One aspect is to allow the user to define a list of syntax rule numbers which should be ignored during the syntax check. (I don't know when this was added to pycodestyle.)

So, rather than overriding the list of ignores picked up by the pycodestyle module I would suggest merging the mu internal hard coded ignore list with the ignore list picked up by the pycodestyle configuration mechanism. 

 The configuration mechanism is detailed at:
      http://pycodestyle.pycqa.org/en/latest/intro.html#configuration

This would allow the user to add their own syntax rules to the list to be ignored when the check is run. So, for the kids we can relax those rules that do not cause functional errors, but still keep the ability to
enforce better coding standards at a later date.

The downside with my proposed fix incorporated:

If the user does "not" specify a custom configuration file then pycodestyle has a default ignore list that is loaded and merged instead. The current default list contains all of the rules as currently defined in 
the mu hard coded list, plus two extra,
    W504: which looks to be "line break after binary operator"
    E24:  which looks to include "241: multiple spaces after ‘,’" and "242: tab after ‘,’"

mu's current ignore list
        ('E121', 'E123', 'E126', 'E226', 'E302', 'E305', 'E24', 'E704', 'W291', 'W292', 'W293', 'W391', 'W503', )

pycodestyle current default ignore list
        ('E121', 'E123', 'E126', 'E226', 'E24', 'E704', 'W503', 'W504')

The net result is if you do not specify a custom configuration file you end up with a list that contains duplicate rules. Adding the calls to set() and tuple() code as shown in the proposed changes below will eliminate the duplicates within the resulting combined tuple.
             style.options.ignore = tuple(set(ignore))

A summary of the original, current, and proposed code changes is below:

The original pre issue #253 code was:
    # Configure which PEP8 rules to ignore.
    style = StyleGuide(parse_argv=False, config_file=False)
    checker = Checker(code_filename, options=style.options)

The code after #253: (note the override of the StyleGuides ignore list)
    # Configure which PEP8 rules to ignore.
    ignore = ('E121', 'E123', 'E126', 'E226', 'E302', 'E305', 'E24', 'E704', 'W291', 'W292', 'W293', 'W391', 'W503', )
    style = StyleGuide(parse_argv=False, config_file=False)
    style.options.ignore = ignore
    checker = Checker(code_filename, options=style.options)

        Commit data for the change supporting request #253:
        commit 2518642c2dde7584ce1b2a78a7383409d9454f2e
        Author: Nicholas H.Tollervey <ntoll@ntoll.org>
        Date:   Fri Sep 15 12:52:20 2017 +0100
            Sane PEP8 error checking.

Proposed update to merge the ignore lists:
    # Configure which PEP8 rules to ignore.
    ignore = ('E121', 'E123', 'E126', 'E226', 'E302', 'E305', 'E24', 'E704', 'W291', 'W292', 'W293', 'W391', 'W503', )
    style = StyleGuide(parse_argv=False, config_file=False)
    # merge in pycodestyle module's user defined or default ignore list 
    ignore = style.options.ignore + ignore
    # remove duplicates with set(), convert back to tuple()
    style.options.ignore = tuple(set(ignore))
    checker = Checker(code_filename, options=style.options)

Steps to test the change:
  1. Open mu and add a test program
  2. add a comment to the test program that does not contain a space after the "#" such as:
       #This is a comment with no space
  3. Run the syntax check via mu's "Check" button
  4. you should see a syntax error reported
     "Block comment should start with '# '
  5. add an environment variable, per your operating system, which points
     to the "directory" containing your custom configuration file:
     From the pycodestyle documentation:

         At the user level, configuration settings are read from the
         following locations:
         If on Windows:
             ~\.pycodestyle
         Otherwise, if the XDG_CONFIG_HOME environment variable is defined:
             XDG_CONFIG_HOME/pycodestyle
         Else if XDG_CONFIG_HOME is not defined:
             ~/.config/pycodestyle

     For example on Linux:
        export XDG_CONFIG_HOME=/home/user1/mu_config

   6. within the configuration directory add a file called "pycodestyle"
   7. Add to the file a section for pycodestyle as follows: This example will case error number 265 to be ignored.
       [pycodestyle]
       ignore = E265

   8 restart the mu editor and open up the test program
   9. Click on the "Check" button. The error noted in step 4  should no longer be reported.

Background:
I'm a long time coder, newbie to python and mu. I too am teaching kids the micro python basics with BBC Microbit. The above was debugged and tested on 14.04.1 Ubuntu Linux using BBC Microbit only.
mu-editor version 1.0.1
Python version 3.6.0
pycodestyle.py version 2.4.0

ntoll, 2019-03-18:
Hi @leroyle, thanks for such a detailed issue. I don't suppose you could put your proposed changes into a PR please..? Thank you!

leroyle, 2019-03-18:
Sure I can see about doing a PR. It will be my first in the open source world but I'll check into the contrib to Mu info. Should not be a problem.

Leroy

MvGulik, 2020-04-20:
Bump.

Is this, potential customizing of the syntax check errors/warnings, already possible in some way?

As Tabs-indentation-only writer I would love to disable some of them. (_starting with my favourite W191 of course._ ;-) )
I did try to set some general/global flake8/pycodestyle configurations. And also tried adjusting mu' s setup.cfg source-file (_as it included a [flake8] section_). But so far I had no success in disabling any error/warning message.

(_mu-editor 1.1.0.alpha.2_)
(_Linux Mint 19.3 cinnamon_)

(_some features (like REPL & Plotter) of the mu-editor are kinda new to me, and I think mu is a nice editor to do some explore of those features. But the lack of being able to adjust the active syntax checks is somewhat of a show stopper (at least for me as dedicated Tab indenter)_)

leroyle, 2020-04-21:
MvGulik,

It looks like the change should be there in 1.1.0.alpha.2, at least according to the change log. I've not installed that version as I'm off on other things these days.

If your on linux, and it looks like you are, the over rides should be set in your home directory, .config subdirectory in a file named pycodestyle.

~/.config/pycodestyle

 which should contain the codes to be overridden such as in this case error 265, warnings 292, 191

[pycodestyle]
ignore = E265, W292, W191

Hope this helps.

  Leroy

MvGulik, 2020-04-21:
Aha ... 
Thanks. Got it to work. :-)

Although I initially did try it with a `pycodestyle` file in the `.config` folder, it did not seem to take.
(_not sure why, and unable to backtrack this. Other than that I found the pycodestyle doc a bit unclear on its folder vs file specifications)_).
So I switched to trying other stuff like `~/.config/pycodestyle/setup.cfg`, which also meant needing to renaming the initial `pycodestyle` file (_ie: disabling the correct way without knowing it (oops)_).

leroyle, 2020-04-21:
Good to see you have it working. 

In case anyone else comes this way, this is from the pycodestyle docs re the location of the user override file. "ignore=" is where the Exxx or Wxxx error/warning numbers would go:

    The behaviour may be configured at two levels, the user and project levels.

    At the user level, settings are read from the following locations:

    If on Windows:
    ~\.pycodestyle

    Otherwise, if the XDG_CONFIG_HOME environment variable is defined:
    XDG_CONFIG_HOME/pycodestyle

    Else if XDG_CONFIG_HOME is not defined:
     ~/.config/pycodestyle
    Example:

[pycodestyle]
count = False
ignore = E226,E302,E41
max-line-length = 160
statistics = True

    At the project level, a setup.cfg file or a tox.ini file is read if present. If none of these files have a 
    [pycodestyle] section, no project specific configuration is loaded._

closed 2020-04-21 by leroyle