mu-editor/mu #784: Latest version doesn't work on WINDOWS 10

johncblacker, 2019-03-07:
Installed latest version on Win 10, installed for all users into "Program Files" and found that I got an error when running.  Pasted the output below:
----------------------------------- output from mu run -------------------------------------------------
c:\Users\jobla\AppData\Local\Programs\Python\Python37-32>python.exe "c:\program files\mu\mu.pyw"
Traceback (most recent call last):
File "c:\program files\mu\mu.pyw", line 29, in <module>
from mu.app import run
File "c:\program files\mu\pkgs\mu\app.py", line 29, in <module>
from PyQt5.QtCore import QTimer, Qt
ImportError: DLL load failed: %1 is not a valid Win32 application.

c:\Users\jobla\AppData\Local\Programs\Python\Python37-32>python.exe "c:\program files\mu\mu.launch.pyw"
Traceback (most recent call last):
File "c:\program files\mu\mu.launch.pyw", line 31, in <module>
from mu.app import run
File "c:\program files\mu\pkgs\mu\app.py", line 29, in <module>
from PyQt5.QtCore import QTimer, Qt
ImportError: DLL load failed: %1 is not a valid Win32 application.
---------------------------- end of pasted log output -------------------------------
Note that I have Python 2.7.9, 3.6.0, 3.7.0 and QT 5.9.3 installed on my computer.
I tried running two different ways as documented above.
Thanks.

carlosperate, 2019-03-07:
Hi @johncblacker,

How did you install Mu? Did you download and execute one of the .exe files from https://codewith.mu/en/download?

You are running your local Python installation for that output, but Mu packs its own Python. What happens if you double click the Mu shortcut?

johncblacker, 2019-03-07:
Yes, I did. Executed mu-editor_1.0.2_win64.exe

johncblacker, 2019-03-07:
No shortcut was created, nor does it show up in my list of all apps.

johncblacker, 2019-03-08:
Well, I went over to the regular win10 start panel and looked there for the mu app and found it, put it on my taskbar executed and it looks to be working!  I guess I'm ok now.  Thanks for the clue about using the shortcut.

closed 2019-03-08 by johncblacker

carlosperate, 2019-03-08:
Glad it is working!