mu-editor/mu #785: Swiss german translation (de_CH)

r0the, 2019-03-08:
Hi
As mentioned in the discussion, here's a Swiss German (de_CH) translation for Mu. It's based on the German (de_DE) translation. I adapted it to Swiss German spelling (ss instead of ß). I also changed some translations which felt unnatural.

[mu.po.zip](https://github.com/mu-editor/mu/files/2946047/mu.po.zip)

tjguk, 2019-03-08:
(for reference this started as a discussion on Gitter somewhere around here https://gitter.im/mu-editor/general?at=5c820c067851db1349889cd7)

wayoda, 2019-05-07:
@r0the 
I fixed a few typos in the German (de_DE) translation for Mu a while ago so I looked at your po-file out of curiosity. Since I only wanted to fix some obvious typos in my PR I didn't want make too many changes to the original translation. 
But the changes you proposed really make sense to me 
* Using **first person singular** when adressing the user feels much nicer
* I like your use of hyphens which improves readability `Reset-Knopf` vs `Reset Knopf`

Since the next release will add some new features the po-files will have to be updated anyway. I will prepare a PR for the de_DE locale which reflects your changes. 
And while we're at it: the new `Tidy` button is giving me headaches. `Säubern` doesn't sound right to me but I really can't come up with a  **catchy** label for the new action. Any ideas?

Grüße
Eberhard

carlosperate, 2019-05-07:
> I will prepare a PR for the de_DE locale which reflects your changes.

Thanks @wayoda!
@r0the if you could help with the PR review that would be fantastic as well.

r0the, 2019-05-08:
@wayoda Thanks for your feedback. It's great that you want to change the de_DE locale accordingly. I think it's best that the two locales don't differ too much.

How about translating `Tidy` to `Aufräumen`?

Grüsse
Stefan

r0the, 2019-05-08:
@carlosperate I'm happy to help with the PR review.

carlosperate, 2021-12-27:
Hi @r0the, i's been a while, but if you are happy to provide translations the Mu project is finally ready again 🎉 

More info in:
- https://github.com/mu-editor/mu/discussions/1931