mu-editor/mu #786: UI/UX: Accidental breakpoint toggling

tmontes, 2019-03-08:
## Context

Earlier this week I delivered a programming introduction workshop for kids, aged 11-12: with Python and Mu, of course! :) We all had a lot of fun and the kids did indeed learn a lot! :)

## The Issue

One surprising observation I noted was the fact that *all of the four* kids consistently used the mouse/trackpad to position the text editing cursor (instead of the arrow keys on their keyboards) when they wanted to "go somewhere" in their code to change it. In retrospect, this probably makes sense given that they are used to touch oriented devices like modern phones and tablets, on one hand, and that their experience in using computer keyboards is associated to simpler, non-multi-line-editing, text input, on the other -- say, submitting a typed in phrase on YouTube to find videos of their liking. :)

As a result of this, and of non-100% accurate pointing and clicking, they very often ended up inadvertedly triggering a "breakpoint toggle action" when trying to place the text editing cursor right at the start of a given line, to the left of the first column -- or, worse, triggering the "Cannot set breakpoint" notification dialog, if the target line wasn't "breakpointable",

Of course, that led to some frustration in the editing process and questions about what that was. Even though I do believe that having beginners running code through a debugger can be a very valuable learning experience, I'd rather introduce the concepts when I plan to do so (which I didn't, given the short amount of time we had), instead of being somewhat "forced" into a dismissive answer like "that's a different thing we may learn about some other day, try not to worry about it -- oh, by the way, have you tried using the keyboard instead?"... :(

## The Proposal

I figured that a solution to minimize such a usability issue would be to increase the "white space" to the left of the first column, but within the "white editing area" (not on the left margin where the line numbers and breakpoint markers are displayed) -- in HTML/CSS speak, setting a left-padding on the "white editable area" of the editor.

Then I went exploring on how to attain that. After nearly two hours I gave up... Looking at Scintilla's source, that spacing seems to be fixed and hard coded. :(

A solution I found that can be implemented, works the other way around, by adding a "right-padding" to the "gray line number and breakpoints marker left margin". I'll be sharing a PR with that, for evaluation.

## The Question

I wonder if others have observed beginner kids do the same thing: using the mouse/trackpad to position the text editing cursor, instead of the arrow keys on the keyboard. More, if you observed that, did you notice kids accidentally triggering breakpoint toggles? Or are those kids "mouse pointing" experts, even if Python beginners? :)

Would you consider the case I'm describing a usability issue? Hopefully yes... Happy to discuss.

Thanks for taking the time in thinking about this and sharing your experiences.

tmontes, 2019-03-08:
PR #787 submitted, along with implementation notes. :)

ntoll, 2019-03-18:
Hi @tmontes, sorry for the delay in getting to this. I'll take a look at the related PR with a view to getting it into this morning's Alpha 1.

However, my (limited) experience of the accidental breakpoint triggering you mention is that it's not been a problem so far. You're the first person to raise this issue, and in the various interviews I've done with teachers about Mu related problems, none have mentioned it. Nevertheless, that doesn't mean a "solution" can't be provided as a guard, for all the reasons you mention above.

I've had a look at the related PR and tested the change... I think it's nice and simple and does exactly what we want. Once Travis CI is passing for OSX I'll merge into master and it'll be in today's alpha 1 release. :-)

closed 2019-03-18 by ntoll

tjguk, 2019-03-18:
As to that: I have seen youngsters accidentally generate a breakpoint in this way. But since it doesn't actually do anything (apart from show a red dot) they usually just shrug and move on. If they ask me I just tell them to click it again and it disappears. And then they shrug and move on. (Can you tell I mostly work with teenage boys?)