mu-editor/mu #788: Add builtins to address Undefined Names

rdagger, 2019-03-10:
Would it be possible to allow adding terms to the linter? For example, Mu flags the MicroPython _const()_ method as 'undefined name'.

In Atom or VS Code, I use Flake8 which allows me to create a config file.  Adding builtins = const to the config file prevents warnings regarding _const()_.

ntoll, 2019-03-10:
Good catch. Yes. This should be a thing. :-)

dybber, 2020-09-29:
An alternative option might be to just add it for the specific modes, (micro:bit and ESP MicroPython modes), or do you see other use-cases for this?

carlosperate, 2020-12-12:
For v1.1 let's add mode-specific keywords, and if we want to expose these via settings file we can do that for v1.2.

devdanzin, 2020-12-17:
The keyword feature is already present, but:
* `const` wasn't added to MicroPythonMode;
* regex handling of flake messages regressed just like in #1161.

The trivial patch below fixes const. Any pointers to other content that should be added to any modes? We should also add a test that runs a real check so we spot any regressions, ` test_check_flake_with_builtins` mocks flake's message so it didn't catch it.

```diff
diff --git a/mu/logic.py b/mu/logic.py
index 0f8edcb..356b2ff 100644
--- a/mu/logic.py
+++ b/mu/logic.py
@@ -460,7 +460,7 @@ def check_flake(filename, code, builtins=None):
     check(code, filename, reporter)
     if builtins:
         builtins_regex = re.compile(
-            r"^undefined name '(" + "|".join(builtins) + r")'"
+            r"^.*\d undefined name '(" + "|".join(builtins) + r")'"
         )
     feedback = {}
     for log in reporter.log:
diff --git a/mu/modes/base.py b/mu/modes/base.py
index 29a5e0e..8e96c9a 100644
--- a/mu/modes/base.py
+++ b/mu/modes/base.py
@@ -384,6 +384,7 @@ class MicroPythonMode(BaseMode):
     force_interrupt = True
     connection = None
     baudrate = 115200
+    builtins = ['const']
 
     def compatible_board(self, port):
         """
```

devdanzin, 2020-12-18:
The builtins_regex should also become a module-level constant, probably in config.py.

PR #1205 (2021-01-26): Update check_flake regex for handling builtins and add "const" builtin, fixes #788
Fix #788 by adding a "const" builtin to MicroPythonMode. For the fix to work, solve a issue similar to  #1161, as our regexes were missing flake's new output formatting and not matching builtins. Moved the regex to a module constant so it's easier to spot it when fixing other regexes.

Updated existing test and added test using real check_flake output to catch any regressions.
commit 3c9f9944 (2021-01-26): Merge pull request #1205 from devdanzin/fix_builtins_regex

Update check_flake regex for handling builtins and add "const" builtin, fixes #788
closed 2021-01-26 by dybber