mu-editor/mu #789: Pyinstaller package mu-editor to exe, in python mode, click Run  button will be restart window.

AustinKuture, 2019-03-15:
Pyinstaller(3.4) package mu-editor to exe, in python mode, click Run  button will be restart window.

ntoll, 2019-03-15:
We don't support PyInstaller. Please use the installers found here: https://codewith.mu/en/download

The problem you report is due to a limitation/design decision in PyInstaller.

ntoll, 2019-03-15:
Out of interest, why were you using PyInstaller to package Mu..?

Full details of how we package Mu can be found here: https://mu.readthedocs.io/en/latest/packaging.html

If you have any questions, please don't hesitate to ask! :-)

closed 2019-03-15 by ntoll

AustinKuture, 2019-03-15:
> 
> 
> Out of interest, why were you using PyInstaller to package Mu..?
> 
> Full details of how we package Mu can be found here: https://mu.readthedocs.io/en/latest/packaging.html
> 
> If you have any questions, please don't hesitate to ask! :-)

Because I design a new UI for children, So, I using PyInstaller to package Mu again

ntoll, 2019-03-15:
Cool. My advice is not to use PyInstaller. The Makefile for Mu already includes commands for building installers for Windows (`make win64`).

Care to share your work on the new UI? I'd love to know what you've changed and why. It's good to share work, progress and research..! :-)

AustinKuture, 2019-03-15:
Thank you very much for your reply.
I wrote a new module that supports ESP32,(Micropython)， want to contribute to other developers use and a neat UI for kids using Python, Pygame

ntoll, 2019-03-15:
@AustinKuture we already have a new mode for ESP8266/32 boards..! You can see the PR for it here: #676

AustinKuture, 2019-03-15:
Oh! Very pity, ESP32/ESP8266 Version of support, I was written in 27  August  2018