mu-editor/mu #791: Third party app installation DOES NOT work on Windows

ntoll, 2019-03-18:
I see the following output when I try to install the module `arrr` in the latest build (see: http://mu-builds.s3-website.eu-west-2.amazonaws.com/?prefix=windows/ build `5576713` )

```
Collecting arrr
  Using cached https://files.pythonhosted.org/packages/6d/ab/c07f37ec7d7eac559c338d44019abd065bf25e06ccd410420907115875ad/arrr-1.0.0b3.tar.gz
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "C:\Users\ntoll\AppData\Local\Temp\pip-install-8cjikf24\arrr\setup.py", line 3, in <module>
        from arrr import get_version
    ModuleNotFoundError: No module named 'arrr'
    
    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in C:\Users\ntoll\AppData\Local\Temp\pip-install-8cjikf24\arrr\

FINISHED
```

The relevant parts of the log file are as follows:

```
2019-03-18 11:13:12,981 - mu.logic:1151(sync_package_state) INFO: Synchronize package states...
2019-03-18 11:13:12,981 - mu.logic:1152(sync_package_state) INFO: Old: set()
2019-03-18 11:13:12,981 - mu.logic:1153(sync_package_state) INFO: New: {'arrr'}
2019-03-18 11:13:12,981 - mu.logic:1157(sync_package_state) INFO: To add: {'arrr'}
2019-03-18 11:13:12,981 - mu.logic:1158(sync_package_state) INFO: To remove: set()
2019-03-18 11:13:12,981 - mu.logic:1159(sync_package_state) INFO: Site packages: C:\Users\ntoll\AppData\Local\python\mu\site-packages
2019-03-18 11:13:12,997 - mu.interface.dialogs:420(run_pip) INFO: C:\Program Files\Mu\Python\pythonw.exe -m pip install arrr --target C:\Users\ntoll\AppData\Local\python\mu\site-packages
```

I really don't know what's going on here. Any help would be most appreciated. Could it be because Python is being run in isolated mode..? Are we using the right Python executable? Is there a missing module that isn't bundled with the installer?

Who knows... (I certainly don't). :-(

tjguk, 2019-03-18:
@ntoll to save me trying: is this only the bundled build which fails? Or does it also fail running from a git clone?

ntoll, 2019-03-18:
Hi @tjguk, thanks for the prompt reply.

This is behaviour from Mu via the installer.

I assume when you tested the third party installer work it was done via your Git clone (using your system wide Python in a virtualenv)..? Looking at the docs(https://docs.python.org/3.4/whatsnew/3.4.html#whatsnew-isolated-mode) I believe isolated mode may be the root of this problem. :-(

My current Windows machine does not have Python installed on it so I'm certain that anything Pythonic in Mu is isolated to Mu. I'm in meetings all afternoon so won't be able to check if it fails with a git clone.

tjguk, 2019-03-18:
Thanks for the info. I'll have a look

ntoll, 2019-03-18:
FWIW, when I run the following command from `C:\Program Files\Mu\Python`, from the terminal, I get the same failure as reported in Mu:

```
python.exe -m pip install arrr --target C:\Users\ntoll\AppData\Local\python\mu\site-packages
```

ntoll, 2019-03-18:
Hi Tim, I've been looking into this a little bit.

The core issues:

* When `pip` is working, at some point, it issues the command: `python setup.py egg_info` **in** the TEMPORARY directory in which the package has been unzipped.
* The temporary directory looks something like this: `C:\Users\ntoll\AppData\Local\Temp\pip-install-1t4qdskg\arrr`.
* Since Mu's Python, which we're using to drive `pip`, is always run in isolated mode then `.` is not on the Python path.
* In a previous ticket we added a `mu.pth` file to the correct location in order to add the desired paths to the Python path. For the purposes of testing, I copied the content of this file and made sure it existed in the right location (e.g. in `c:\Users\ntoll\AppData\Roaming\Python\Python36\site-packages`)
* Here's the problem... the TEMPORARY directory is not in the `mu.pth` file, hence the import error when `setup.py` tries to import `arrr.py`.
* Since this is `pip` creating the temporary directory, Mu cannot know in advance where this directory may be, so it cannot be added to the `.pth` file.

Question: I wonder if there's any way to specify the temporary directory used by `pip` to unpack packages, or override this is some way? The alternative is some way to force Python to use `.` as being on the path.

Answering my own question... `pip install --help` says:

```
  -b, --build <dir>           Directory to unpack packages into and build in.
                              Note that an initial build still takes place in
                              a temporary directory. The location of temporary
                              directories can be controlled by setting the
                              TMPDIR environment variable (TEMP on Windows)
                              appropriately. When passed, build directories
                              are not cleaned in case of failures.
```

I've not yet managed to play around with this possibility, and I'm about to step away from my computer to get home. My instinct would be to make sure we specify a path to a directory which is also in `mu.pth`.

My concern is the resulting TMPDIR may end up being filled with subdirectories, one for each of the to-be-installed packages, and each of which will need to be added to the Python path in some as-yet-to-be determined manner. Also, what happens when there's a package with dependencies which may not be known in advance, thus meaning there will be subdirectories in the temporary directory which Mu can't foresee require adding to the Python path..? Ugh.

FWIW, the `run_pip` method in `mu.interface.dialog.PackageDialog` is where we specify the args to be passed into `pip`.

There are also these options which look interesting.

```
  --no-build-isolation        Disable isolation when building a modern source
                              distribution. Build dependencies specified by
                              PEP 518 must be already installed if this option
                              is used.
  --install-option <options>  Extra arguments to be supplied to the setup.py
                              install command (use like --install-option="--
                              install-scripts=/usr/local/bin"). Use multiple
                              --install-option options to pass multiple
                              options to setup.py install. If you are using an
                              option with a directory path, be sure to use
                              absolute path.
  --global-option <options>   Extra global options to be supplied to the
                              setup.py call before the install command.
```

...although it's hard to tell from looking at the docs for `setup.py` what useful options one could possibly pass in here -- hopefully adding the `CWD` to the Python path might be possible, or perhaps using `--develop` as a flag into `setup.py` will give us some more flexible options. Nor am I convinced that `--no-build-isolation` would override the fact that we're using Mu's Python to drive pip and it's running in isolated mode already.

Right..! That's my "braindump" done. Now you know what I know. :-) Any help with this would be **greatly** appreciated.

ntoll, 2019-03-18:
After reading a bit more (no actual testing done here), I wonder if this approach might bodge a work-around (this makes me very uncomfortable):

https://stackoverflow.com/a/49405161

ncoghlan, 2019-03-19:
Note that the Python interpreter's execution isolation (i.e. ignore env vars, script dir not on sys.path) isn't the same thing as pip's build isolation (hide the target install environment from the build environment).

You're correct that it is the interpreter level isolated mode that will keep setup.py from appearing on sys.path, though. The best way to add it back would be to run the build with "-Es" (ignore environment, ignore user site packages) rather than coming up with a more convoluted way of achieving same thing (if you run Python with "-I", and then arrange to put the script directory on sys.path anyway, all you've done is express exactly what "-Es" does in a more convoluted way)

tjguk, 2019-03-19:
Thanks @ncoghlan. The funny thing is that I can't actually reproduce this effect by running pip via a `python -I` command line, but I suspect that's because pip shells out to run `setuptools egg_info` by calling sys.executable (and then `-c`-ing a command to unwrap and exec `setup.py`) without passing along any existing commandline parameters. In our case, I *think* (but I'm not sure) that the isolation is somehow built into the python executable we're using via pynsist.

ntoll, 2019-03-19:
Hi folks, just catching up after a long day of (CodeGrades) related meetings and my evening is about to turn into a dad-taxi armageddon. ;-)

@ncoghlan thanks for the heads up, this gives me more information to work on and I'll try fiddling with various settings and report back here with the results. Good stuff! :-)

@tjguk I spent about 30 minutes pre-today's meetings trying out some hunches. I managed to "make it work" (note the quotes) by adding known paths to the `mu.pth` file we drop to the right place (not an ideal situation, and I hope Nick's suggestions get us to a better place). However, this only led to a brand new and interesting error message:

```
C:\Program Files\Mu>Python\python.exe -m pip install --build=C:\Users\ntoll\AppData\Local\Temp --no-cache-dir --target C:\Users\ntoll\AppData\Local\python\mu\site-packages arrr
Collecting arrr
  Downloading https://files.pythonhosted.org/packages/6d/ab/c07f37ec7d7eac559c338d44019abd065bf25e06ccd410420907115875ad/arrr-1.0.0b3.tar.gz
    Complete output from command python setup.py egg_info:
    running egg_info
    creating pip-egg-info\arrr.egg-info
    writing pip-egg-info\arrr.egg-info\PKG-INFO
    writing dependency_links to pip-egg-info\arrr.egg-info\dependency_links.txt
    writing entry points to pip-egg-info\arrr.egg-info\entry_points.txt
    writing top-level names to pip-egg-info\arrr.egg-info\top_level.txt
    writing manifest file 'pip-egg-info\arrr.egg-info\SOURCES.txt'
    C:\Program Files\Mu\pkgs\setuptools\dist.py:475: UserWarning: Normalizing '1.0.0.beta.3' to '1.0.0b3'
      normalized_version,
    error: [Errno 2] No such file or directory: 'C:\\Program Files\\Mu\\Python\\python36.zip\\lib2to3\\Grammar.txt'

    ----------------------------------------
```

Googling around the error message led me to the record of a conversation involving a friendly and helpful Australian dude called Nick. ;-) Turns out we there is a solution for this error (believe it or not, unzip the contents of `python36.zip` into a directory called `python36.zip`). Hurrah for computers... ;-)

I intend to focus on this tomorrow morning..! I'll let you know how I get on.

ntoll, 2019-03-22:
Hi @tjguk @ncoghlan,

A quick heads up on how I've resolved this.

* Python was running in isolated mode, not because of the `-I` flag but because the embeddable version of Python we were using had a `pythonXY._pth` in the same directory as the `pythonXY.exe` file. Removing the file fixed this problem.
* While the path issue was solved, I ran into the `"No such file or directory: "C:\Program Files\Mu\Python\python36.zip\lib2to3\Grammar.txt"` problem, which is fixed only if the file `pythonXY.zip` is actually an unzipped directory called `pythonXY.zip`.
* Because of the requirement requested by school network admins that Mu work in a way that doesn't interfere with other installed apps, I've modified the shortcut which launches Mu to use the `-Es` flags.

While these three changes are, in themselves, quite minor, working through the problem to identify that it was *just* these three simple changes that needed to happen was a bit of a "process".

In any case, thank you for your help with this (yes, you helped). The vague memory of how Mu is started with isolated Python led me to the `_pth` file which was the key log to unblock in order to find the solution. :-)

closed 2019-03-22 by ntoll

tmontes, 2019-03-22:
@ntoll, is there any PR / commit with the changes you just mentioned? I can't seem to be finding any... :)

ntoll, 2019-03-22:
@tmontes that's my fault. I (mistakenly) did the work on `master` and pushed on auto-pilot before I realised what I'd done. The most recent handful of commits will reveal all. For example, the changes to the Windows packaging happen here: e20976fafdb

Apologies for my cock-up.

tmontes, 2019-03-22:
@ntoll thanks: I'll have to have a look at that, such that the improvements/fixes in the Windows building process are properly integrated/propagated into PR #779. Not sure when, as I've been pretty busy lately. :)