mu-editor/mu #792: Error using debug on MacOS Mojave

TommyPujol06, 2019-03-19:
I just installed the Mu editor on Mojave and I created a test file to see if there are differences between using Mu in Linux and Mac. I pressed on Debug (didn't mean to) and then it said : 
>Overriding built-in editor font due to Issue #552

Then it does nothing.
[image]
Mojave version : 10.14.3

ntoll, 2019-03-19:
Hi @TommyPujol06, thanks for reporting this.

Calling @stestagg. Any thoughts on this..? See our conversation in #552.

stestagg, 2019-03-19:
Hi @TommyPujol06 

Thanks for this!  It's definitely a weird user experience.

I think what's happening is that functionally, mu is behaving correctly.  When you click on 'Debug', mu pauses on the first executable line, and waits for you to click Continue (or something else).  This is indicated by the yellow background on line 2.

On Windows, you can see this behaviour:
[image]

On OSX, for some reason the 'Overriding...' message is being shown in the console when the debugger starts up (I'm looking into this). and that is drawing the attention away from the debugging buttons above, so it's not clear that Mu is waiting for input to proceed.

I think ideally, we'd output some text to the console in all cases when debug is pressed, to make it clear that mu will pause before the first executable statement.  But we should also stop that font message from being displayed.

carlosperate, 2019-03-19:
Just for reference: https://github.com/mu-editor/mu/issues/751

closed 2019-03-20 by TommyPujol06