mu-editor/mu #795: When an opened file is deleted, opening a different file crashes Mu

carlosperate, 2019-03-21:
To replicate:
* Create a new file, e.g. `to_be_deleted.py`
* Open `to_be_deleted.py` with Mu
* Delete `to_be_deleted.py` from your computer
* Click Load on Mu and open a different file
* Mu will crash

```
2019-03-21 10:38:39,430 - mu.logic:811(get_dialog_directory) INFO: Using path for file dialog: /Users/microbit-carlos/mu_code
2019-03-21 10:38:40,973 - mu.interface.main:251(get_load_path) DEBUG: Getting load path: /Users/microbit-carlos/mu_code/one_chunk_minus.py
2019-03-21 10:38:40,973 - mu.logic:715(_load) INFO: Loading script from: /Users/microbit-carlos/mu_code/one_chunk_minus.py
2019-03-21 10:38:40,973 - root:95(excepthook) ERROR: Unrecoverable error
Traceback (most recent call last):
  File "/Users/microbit-carlos/Downloads/micropython/mu-editor-1.0.2.app/Contents/Resources/app/mu/logic.py", line 831, in load
    self._load(path)
  File "/Users/microbit-carlos/Downloads/micropython/mu-editor-1.0.2.app/Contents/Resources/app/mu/logic.py", line 730, in _load
    if os.path.samefile(path, widget.path):
  File "/Users/travis/build/mu-editor/mu_portable_python_macos/python/lib/python3.6/genericpath.py", line 97, in samefile
FileNotFoundError: [Errno 2] No such file or directory: '/Users/microbit-carlos/mu_code/to_be_deleted.py'
2019-03-21 10:39:50,134 - root:112(run) INFO: 
```

ntoll, 2019-03-21:
@carlosperate have you tested this with `master` branch. I'm pretty certain I fixed something like this a couple of weeks ago.

carlosperate, 2019-03-21:
No, I haven't tested with master, I'll do that now.

carlosperate, 2019-03-21:
I can confirm this is fixed in master, sorry!

closed 2019-03-21 by carlosperate