mu-editor/mu #808: Copy-Paste via keyboard inconsistent in REPL between microbit mode and python mode

askvictor, 2019-04-02:
Copying or pasting in the REPL works fine via the context menu, but there is different behaviour using keys (e.g. Ctrl-C, Ctrl-V) depending on the mode. On Windows, in Python 3 mode, Ctrl-C and Ctrl-V work for copy and paste. In Micro:bit mode, these do not work, but rather Ctrl-Shift-C and Ctrl-Shift-V work. 

I'm uncertain how this looks on a Mac which uses Cmd-C/V.

Related (closed) issue: https://github.com/mu-editor/mu/issues/169

nealmcb, 2019-09-22:
I can confirm this, on Ubuntu Linux. Very confusing!!

dybber, 2020-02-15:
When interacting with the MicroPython REPL, Ctrl-C sends an interrupt/abort signal that stops any current process. Similarly Ctrl-D does a soft-reboot.

To be able to use Ctrl-C for that purpose, Ctrl-Shift-C is used for copy-paste instead, and to mirror that Ctrl-Shift-V is used for pasting. I don't see any good alternative solutions, but chip in if you have any suggestions for alternative behavior. One possibility would be an explicit "Abort" button in the UI, but then you would be unable to teach what Ctrl-C interrupt does.

devdanzin, 2020-12-26:
Could we maybe raise a "Confirm Abort" dialog on Ctrl-C, so it teaches both shortcuts and avoids aborts when user wanted to copy? Confirmation could then be by button click and/or hitting Ctrl-C again, so users don't break their flow (assuming sending Ctrl-C twice when not in Mu isn't a bad habit).

street-grease-coder, 2020-12-31:
The expected behavior would be: if text is selected: copy text, if not text is selected, abort. I don't know how MU works internally - but if the console does any processing before sending commands to the microcontroller, this is where this logic would go. In the same category, pressing up for recent command and delete sometimes (using backslash) also sometimes gives weird results, it even may look right but you seemingly end up sending another command.

dybber, 2021-01-02:
What is the normal behavior in a Windows command prompt/Powershell, when you push CTRL-C? I'm on Mac, so we don't have this issue (CTRL-C is abort, CMD-C is copy), so I can't check myself.

In my opinion we should follow standard practice and teach the users how to operate a shell/terminal/command prompt for the OS they are on. We can add pop-ups to teach this, but we shouldn't invent our own "copy if text is selected" in my opinion.

Thus, if the standard practice on Windows is for CTRL-C to send abort, then we should do the same. We could then add a small popup notification that this is what is going to happen (with buttons: cancel, ok, and never show this again), and also explain that copy is on "CTRL-SHIFT-C" or whatever it is on Windows.