mu-editor/mu #809: ModuleNotFoundError when installing 3rd-party package openpyxl in 1.1.0 Alpha 1

bigeyex, 2019-04-02:
I'm using 1.1.0 Alpha 1 and trying to install a 3rd party package named openpyxl
I clicked the gear icon, select "third party package" tab, entered "openpyxl", then clicked OK.
Instead of getting the package installed, it shows:
```
Collecting openpyxl
Collecting et-xmlfile (from openpyxl)
  Using cached https://files.pythonhosted.org/packages/22/28/a99c42aea746e18382ad9fb36f64c1c1f04216f41797f2f0fa567da11388/et_xmlfile-1.0.1.tar.gz
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "C:\Users\95\AppData\Local\Temp\pip-install-eeame9lz\et-xmlfile\setup.py", line 40, in <module>
        from et_xmlfile import (
    ModuleNotFoundError: No module named 'et_xmlfile'
    
    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in C:\Users\95\AppData\Local\Temp\pip-install-eeame9lz\et-xmlfile\

FINISHED
```

I'm working on Windows 10.

Mu Log:
```
-----------------

Starting Mu 1.1.0.alpha.1
2019-04-03 01:14:19,239 - root:113(run) INFO: uname_result(system='Windows', node='95-PC', release='10', version='10.0.17763', machine='AMD64', processor='Intel64 Family 6 Model 60 Stepping 3, GenuineIntel')
2019-04-03 01:14:19,240 - root:114(run) INFO: Python path: ['D:\\Apps\\Mu\\pkgs', 'D:\\Apps\\Mu\\Python\\python36.zip', 'D:\\Apps\\Mu\\Python', 'D:\\Apps\\Mu\\pkgs', 'C:\\Users\\95\\AppData\\Local\\python\\mu\\site-packages', 'D:\\Apps\\Mu\\pkgs\\IPython\\extensions']
2019-04-03 01:14:19,240 - root:115(run) INFO: Language code: zh_CN
2019-04-03 01:14:19,354 - mu.logic:574(__init__) INFO: Setting up editor.
2019-04-03 01:14:19,354 - mu.logic:596(__init__) INFO: Settings path: C:\Users\95\AppData\Local\python\mu\settings.json
2019-04-03 01:14:19,354 - mu.logic:597(__init__) INFO: Session path: C:\Users\95\AppData\Local\python\mu\session.json
2019-04-03 01:14:19,354 - mu.logic:598(__init__) INFO: Log directory: C:\Users\95\AppData\Local\python\mu\Logs
2019-04-03 01:14:19,354 - mu.logic:599(__init__) INFO: Data directory: C:\Users\95\AppData\Local\python\mu
2019-04-03 01:14:19,378 - mu.logic:612(setup) INFO: Available modes: python, adafruit, microbit, esp, debugger, pygamezero
2019-04-03 01:14:19,689 - mu.logic:1225(change_mode) INFO: Workspace directory: C:\Users\95\mu_code
2019-04-03 01:14:19,698 - mu.logic:661(restore_session) INFO: Restoring session from: C:\Users\95\AppData\Local\python\mu\session.json
2019-04-03 01:14:19,698 - mu.logic:662(restore_session) DEBUG: {'theme': 'day', 'mode': 'python', 'paths': ['E:\\workspace\\playground\\py-in-work\\lesson1.py'], 'envars': [], 'minify': False, 'microbit_runtime': '', 'zoom_level': 2}
2019-04-03 01:14:19,699 - mu.logic:749(_load) INFO: Loading script from: E:\workspace\playground\py-in-work\lesson1.py
2019-04-03 01:14:19,699 - mu.logic:305(read_and_decode) DEBUG: Trying to decode with utf-8
2019-04-03 01:14:19,700 - mu.logic:308(read_and_decode) INFO: Decoded with utf-8
2019-04-03 01:14:19,700 - mu.logic:321(read_and_decode) DEBUG: Detected newline '\n'
2019-04-03 01:14:19,700 - mu.logic:832(_load) DEBUG: from openpyxl import Workbook

wb = Workbook()
sheet = wb.active

2019-04-03 01:14:19,974 - mu.logic:683(restore_session) INFO: Loaded files.
2019-04-03 01:14:19,974 - mu.logic:687(restore_session) INFO: User defined environment variables: []
2019-04-03 01:14:19,974 - mu.logic:691(restore_session) INFO: Minify scripts on micro:bit? False
2019-04-03 01:14:19,990 - mu.logic:1225(change_mode) INFO: Workspace directory: C:\Users\95\mu_code
2019-04-03 01:14:23,877 - mu.logic:1120(show_admin) INFO: Showing admin with logs from C:\Users\95\AppData\Local\python\mu\Logs\mu.log
2019-04-03 01:14:23,877 - mu.logic:157(installed_packages) INFO: Packages found: []
2019-04-03 01:14:39,930 - mu.logic:1162(sync_package_state) INFO: Synchronize package states...
2019-04-03 01:14:39,930 - mu.logic:1163(sync_package_state) INFO: Old: set()
2019-04-03 01:14:39,930 - mu.logic:1164(sync_package_state) INFO: New: {'openpyxl'}
2019-04-03 01:14:39,930 - mu.logic:1168(sync_package_state) INFO: To add: {'openpyxl'}
2019-04-03 01:14:39,931 - mu.logic:1169(sync_package_state) INFO: To remove: set()
2019-04-03 01:14:39,931 - mu.logic:1170(sync_package_state) INFO: Site packages: C:\Users\95\AppData\Local\python\mu\site-packages
2019-04-03 01:14:39,932 - mu.interface.dialogs:456(run_pip) INFO: D:\Apps\Mu\Python\pythonw.exe -m pip install openpyxl --target C:\Users\95\AppData\Local\python\mu\site-packages
2019-04-03 01:14:48,949 - mu.logic:1120(show_admin) INFO: Showing admin with logs from C:\Users\95\AppData\Local\python\mu\Logs\mu.log
2019-04-03 01:14:48,949 - mu.logic:157(installed_packages) INFO: Packages found: []
```

bigeyex, 2019-04-04:
This is really weird: when I run from the source, the issue disappeared.
Would anybody confirm that whether this will reproduce with the release package?

tjguk, 2019-04-04:
Thanks for confirming that, @bigeyex. I half-expected that, because our release package has certain restrictions -- although I thought that this one wouldn't engage. I'll try to have a look at some point.

tjguk, 2019-04-07:
Well, not that this really helps but... I'm getting an error doing a simple "pip install openpyxl" into a mu venv. I'll try another module just to make sure I haven't completely messed up the venv. But there may be bigger issues here than the Mu approach...

```
Traceback (most recent call last):
  File "c:\work-in-progress\pip\src\pip\_internal\cli\base_command.py", line 178, in main
    status = self.run(options, args)
  File "c:\work-in-progress\pip\src\pip\_internal\commands\install.py", line 352, in run
    resolver.resolve(requirement_set)
  File "c:\work-in-progress\pip\src\pip\_internal\resolve.py", line 131, in resolve
    self._resolve_one(requirement_set, req)
  File "c:\work-in-progress\pip\src\pip\_internal\resolve.py", line 294, in _resolve_one
    abstract_dist = self._get_abstract_dist_for(req_to_install)
  File "c:\work-in-progress\pip\src\pip\_internal\resolve.py", line 242, in _get_abstract_dist_for
    self.require_hashes
  File "c:\work-in-progress\pip\src\pip\_internal\operations\prepare.py", line 348, in prepare_linked_requirement
    abstract_dist.prep_for_dist(finder, self.build_isolation)
  File "c:\work-in-progress\pip\src\pip\_internal\operations\prepare.py", line 157, in prep_for_dist
    self.req.prepare_metadata()
  File "c:\work-in-progress\pip\src\pip\_internal\req\req_install.py", line 529, in prepare_metadata
    self.run_egg_info()
  File "c:\work-in-progress\pip\src\pip\_internal\req\req_install.py", line 603, in run_egg_info
    print("About to call setup.py with", "\n".join(egg_info_cmd.split(";")))
AttributeError: 'list' object has no attribute 'split'
```

tjguk, 2019-04-07:
Oddly, though, it pip installs ok into a fresh venv. Might be a clash of setuptools versions or something... Oh well: just another strand of yak hair...

tjguk, 2019-04-07:
(Slight apologies for the running commentary...)
Hmmm... Looks like pip had picked up a pre-release version of pip (19.1dev...) somehow... which is concerning in its own right. Re-updated to the latest release (19.0.3) and now able to instal openpyxl in the venv itself.

tjguk, 2019-04-07:
Well, this is unhelpful... I just built an installer from from the current master, installed it in a different user, added openpyxl as a 3rd-party package and ... it worked!

Now this *is* a user account which I generally use for testing mu: I uninstall whatever version is present and then reinstall the latest build. It is possible that there's some kind of residual setup which is helping this to work. So I'll try it in a brand new user account which I'll create for the purpose.

tjguk, 2019-04-07:
Nope: that worked as well. I'm effectively flummoxed at this point. It's clear that the et_xmlfile setup.py is importing from et_xmlfile, assuming that an import in the current directory will work:

https://bitbucket.org/openpyxl/et_xmlfile/src/50973a6de49c9451ec9cdbadd0e4a5a95a2e52b4/setup.py?at=default&fileviewer=file-view-default#setup.py-40

And it's true that, previously, I would have expected that import to fail because we were implicitly disallowing imports from the current directory.

But the new mu_nsis installer wrapper which @ntoll added a few weeks ago should have relaxed that restriction. And my build suggests that it has.

@bigeyex if you haven't, could you try building an installer and running that from the current master, please? if you have, could you confirm when you built?

bigeyex, 2019-04-08:
On my way. Not familiar with the packaging process though.
Here my first computer stops:
[image]

tjguk, 2019-04-08:
Interesting. I got that, too. Looks like we're missing a dependency. For now: `pip install simplegeneric` and then try again. Thanks for persevering :)

ntoll, 2019-04-08:
Hmm... it *was* working a few weeks ago. I think it must be missing from the `requirements.txt` file. That'll be my fault (sorry about that). Let me check what's going on...

ntoll, 2019-04-08:
OK... see 609b6e81db to fix this problem. Once again, sorry about that (missing `simplegeneric` module).

bigeyex, 2019-04-08:
Hmm.. It works on my second computer. Will test on the other 2.

bigeyex, 2019-04-09:
Tested on my second computer. Works fine. I think this issue could be closed. 
What do you think?

tjguk, 2019-04-09:
I'll close it as we have four positive tests (two from you; two from me).

If you do come up with another instance, feel free to comment here as we might have some odd environmental holdover which still affects things...

closed 2019-04-09 by tjguk

ntoll, 2019-04-09:
Thank you both for looking into this issue. It is both frustrating and a bit of a relief that it can't be reproduced. That's computers for you I guess... :-)