mu-editor/mu #810: Adafruit Metro 328 detected as ESP MicroPython Device

makermelissa, 2019-04-03:
This probably isn't a big deal and you can close this issue if you'd like, but I noticed on Mu Alpha 1.1 that the Adafruit Metro 328 comes up as being detected as an ESP MicroPython Device. The memory is probably too small to run MicroPython and it wouldn't really harm anything.

ntoll, 2019-04-03:
@makermelissa aha... I wondered when the first false positive would come up. My guess is the USB controller on the Metro 328 is the same as for an ESP based board, so reporting the same VID/PID..???

dhalbert, 2019-04-03:
Adafruit board detection is now based on the VID (0x239a) only. We thought this was a compromise: some false positives vs. not having to add new boards all the time.

makermelissa, 2019-04-03:
Makes sense.

carlosperate, 2019-04-15:
> Adafruit board detection is now based on the VID (0x239a) only. We thought this was a compromise: some false positives vs. not having to add new boards all the time.

This sounds like a good compromise. The main disadvantage is that Mu will assume all boards with that VID (including future ones) are CircuitPython boards. So there could be false positives, but Mu also assumes the board is running CircuitPython (and in some cases this boards could be running something else, like an Arduino program), so to some extent we are already trusting the user is plugging-in a board that is ready to be programmed with Python.

I think the Metro is based on the CP2014, which some ESP boards use as their UART-to-USB bridge as well. Is Adafruit changing the CP2014 VID to 0x239a as well?

dybber, 2020-09-05:
For the Adafruit Metro 328, I guess this actually doesn't matter that much, as it will probably never be able to run MicroPython/CircuitPython.

At some point however, we should consider making it possible to change modes independently of the selected device. On connection it should still suggest one mode, but if you really want to use another mode instead, I think you should be allowed to try to do that, with appropriate warnings. 

In the future we will probably have many such modes, there's already pull requests for at least 4 such MicroPython based modes (Pyboard, Makeblock Studuino:bit, snek-mode), and they might overlap in device compatibility due to the issue with the same VID/PID being reused by many vendors.

dybber, 2020-09-27:
Actually, that is the case as it is. It might be detected as one type of device, but you can choose another mode and use that. I will close this issue, as mentioned in my last comment the Adafruit Metro 328 will probably never be able to run either MicroPython or CircuitPython.

closed 2020-09-27 by dybber