mu-editor/mu #816: Homepage: Instruction links 404

stonecrusher, 2019-04-15:
Go to https://codewith.mu/en/download
All the instructions are unavailable.
Example: https://codewith.mu/en/howto/install_windows

ntoll, 2019-04-15:
Ooooh... thank you for the heads up. I just fixed it.

The problem was that the path to files now has a version number in it (in preparation for 1.1). Ergo, all the old path no longer work and I missed the links on the download page when I updated.

closed 2019-04-15 by ntoll

stonecrusher, 2019-04-15:
Thank you for the fast fix!

Unfortunately I just realized that there are more faulty links... everything inside the tutorials linking to other tuts :(

Example:
https://codewith.mu/en/tutorials/1.0/logs :
`how to read logs in Mu`
`how to do this` (2x)

Oh and congratulations on your new tuba :D As an orchestral drummer I'm always very close to the tuba guys and they are like our connection to the orchestra (close enough for whispering during rehearsals) and by playing in different orchestras I can appreciate a good tubist now!