mu-editor/mu #818: Feature request - image display

robmarkcole, 2019-04-23:
I really love the live plotting mode, it is great for basic sensor data. However I also have a use case for plotting area date, e.g. from an array sensor such as https://learn.adafruit.com/adafruit-amg8833-8x8-thermal-camera-sensor/python-circuitpython
Can this be added as a feature request?

ZanderBrown, 2019-07-22:
I'm not sure what your asking for?

Do you want different types of graph? If QChart supports it then I can't see any immediate reason why not

robmarkcole, 2019-07-23:
Hi @ZanderBrown 
the data is a 32x32 numpy array being returned by an IR sensor. I would normallu plot it using matplotlib imshow [e.g](https://matplotlib.org/3.1.1/gallery/images_contours_and_fields/image_demo.html#sphx-glr-gallery-images-contours-and-fields-image-demo-py)

aflaxman, 2020-09-02:
I'm interested in this feature request also, and I bet there is a way to do it with QtCharts.  Maybe QPixmap is the way to do it https://pythonbasics.org/pyqt-qpixmap/ or maybe there is a way to link matplotlib into the existing Qt panel https://matplotlib.org/faq/usage_faq.html#how-do-i-select-pyqt4-or-pyside

I can do some experimenting, but it would be great to get some hints.

carlosperate, 2020-09-03:
A specialised plot might be outside of the Mu scope, but if the plot is generic enough and enough people request it we could consider its inclusion if somebody is happy to work on a PR.

In the meantime the Python mode REPL uses IPython, which includes matplotlib, and as Mu comes with PySerial as well it should be possible to create a simple script to get the serial data from devices and plot it with matplotlib.

aflaxman, 2020-09-03:
Oh, nice.  Using matplotlib in the REPL is pretty close to what I want.  I see that if I use matplotlib with Run, it works also, and opens a new window:

```
import numpy as np, matplotlib.pyplot as plt
plt.imshow(np.random.uniform(size=(12,6)))
plt.show()
```

[image]

This is probably just fine for my intended use-case, but I'll circle back to this issue if I need more.

closed 2021-03-24 by carlosperate