mu-editor/mu #820: Installing mu with Python 3.7 on windows - problem.... - already solved.

abixadamj, 2019-04-26:
When I install mu-editor 1.02 on Windows 10, everything is OK ;-) 
Nicholas - great job. 
But - I need some module [https://pypi.org/project/PyTechBrain/](https://pypi.org/project/PyTechBrain/)- this is my own library. So I need to have pip installed - and I need to install Python. And there is a problem - on python.org there is only 3.7 version available to download, and when I install it, I cannot run mu-editor.
It might be because Mu uses Python 3.6 - If someone wants to look: https://vimeo.com/332477215/afa4b696ea

The solution is: When you need Python, get 3.6 version:
https://www.python.org/downloads/release/python-360/

[image]
Adam

abixadamj, 2019-04-26:
This is just for others people - if someone will have the same problem.

closed 2019-04-26 by abixadamj

ntoll, 2019-04-26:
@abixadamj aha... you hadn't mentioned you wanted to install a third party package..! If you use the alpha version of Mu, you'll see that there's a way to install third party packages. :-) See the blog post here for more information: http://madewith.mu/mu/users/2019/03/25/alpha1.html

carlosperate, 2019-04-26:
> So I need to have pip installed - and I need to install Python. And there is a problem - on python.org there is only 3.7 version available to download, and when I install it, I cannot run mu-editor.

It sounds like there is a problem doing `pip install mu-editor` with Python 3.7. This is perhaps an issue we should fix in master and release as v1.0.3 in the meantime before v1.1-final is fully released.

@abixadamj what error did you get?