mu-editor/mu #821: pgzero mode Music button should be relative to code file

tjguk, 2019-04-29:
(Found with @linguini2004)

Running current master in pygame zero mode. Editing code in a folder which is not the default mu_code folder. 

Clicking the [Music] button opens mu_code/music. Copied a file in there, but when the game is run, pgzero reports that the music file cannot be found.

Without looking, my guess is that pgzero looks for the music folder either relative to the file being run or relative to cwd (which in the dev. version of Mu are the same thing). Mu probably assumes relative to mu_code. Certainly creating a music folder under the code folder and copying the file there solved the problem.

Assuming I'm right I do regard this as a bug in Mu. ie I think Mu should open a Music (and probably Images) folder relative to the path of the currently-edited file. There is some possible source of confusion in that clicking the button against different files will yield different folders. But I think that's outweighed by the fact that *not* doing that will produce a difficult-to-diagnose error message from pygame zero.

tmontes, 2019-10-05:
@tjguk, given that PR #862 has been merged, I suppose this issue can be closed, right? :)

tjguk, 2019-10-05:
Yes; thanks. (I'm lousy at back-tracking to close the issues I've PR-ed!)

closed 2019-10-05 by tjguk

tmontes, 2019-10-05:
> Yes; thanks. (I'm lousy at back-tracking to close the issues I've PR-ed!)

...who isn't? That's what computers are good at. :)