mu-editor/mu #826: requirements.txt problem for ARM architecture

askvictor, 2019-05-07:
The requirements.txt includes some lines such as `qscintilla; "arm" not in platform_machine` which exclude the requirements for ARM devices. However, ARM can appear to python as aarch64 (at least my ARM Chromebook running crostini shows this when I call `platform.machine`). Is this exclusion supposed to be for all ARM devices?

carlosperate, 2019-05-07:
Yes, the exclusion is meant for all ARM devices, but because the string returned by `platform.machine` can vary broadly, is going to be difficult to catch all possibilities.

Checking for `arm` being inside the string was just a way that worked in Raspbian with Raspberry Pi 1/Zero (with armv6 processor) and Pi 2 (armv7 and armv8/arm64) and Pi 3 (armv8/arm64). I don't think we have tried this is any other Arm-based platform, so not sure what other strings we can expect apart from `aarch64` or something with `arm` in the title.

Looks like  `aarch64` is also a fairly common string we should be checking as well (https://stackoverflow.com/questions/31851611/differences-between-arm64-and-aarch64) although I'm not sure what would be the best way to do that. We'll have to check if there is an "or" operator available in environmental markers: https://www.python.org/dev/peps/pep-0508/

askvictor, 2019-05-08:
Looks like environmental markers do support "or". I've just tried `"arm" not in platform_machine and "aarch" not in platform_machine` which works fine to exclude those packages on those platforms. Might be worth updating the docs to specify ARM devices rather than just Raspberry Pi

closed 2020-10-04 by carlosperate

carlosperate, 2020-10-13:
@askvictor this was automatically closed with PR https://github.com/mu-editor/mu/pull/1104, but it'd be great if you'd be able to check if it works in chromebooks?